openhwfoundation/core-v-verif #453: Interrupt signal is asserted but mie register is not updated

prachi789, 2021-02-03:
Hi ,

Right now I'm working on interrupt_test (cv32/tests/programs/custom/interrupt_test). 
In that one instruction (csrrs  x0,mie,x12) is generated in .objdump file.
So in GPR ( X12 register value is changed according to logic for generating interrupt) but mie register is not updated.
I've checked waveform also value of MIE register is 0 throughout the simulation.

Thanks in advance.

[image]
[image]
[image]

MikeOpenHWGroup, 2021-02-03:
Hi @prachi789. I am not able to reproduce your issue.

I have modified the testcase to explicitly add a read-back check of MIE after the write and it passes as expected.   Have a look at the `issue453` branch on my personal GitHub repo at https://github.com/MikeOpenHWGroup/core-v-verif/.   I have also looked at a wave-dump after running the test and the MIE is being updated as expected.

I will need additional information to help you further.  Specifically, I would like to have the _exact_ commands you used to clone your working copy of the code and the make command you used to run the test.

Finally, please note that the interrupt_test does not run under the "core" testbench, you must use the "uvmt_cv32" environment.

prachi789, 2021-02-04:
Hi ,

I've cloned an updated repository (
https://github.com/openhwgroup/core-v-verif).

First I've passed below command for generating a hex file.
    make SIMULATOR=vsim interrupt_test

After that I've passed this below command for generating waveforms.
    make SIMULATOR=vsim interrupt-test-gui

Thanks in advance.

MikeOpenHWGroup, 2021-02-04:
Hi @prachi789, I just saw your email on this topic:

> I've cloned an updated repository (https://github.com/openhwgroup/core-v-verif).
>
>First I've passed below command for generating a hex file.
>make SIMULATOR=vsim interrupt_test
>
> After that I've passed this below command for generating waveforms.
> make SIMULATOR=vsim interrupt-test-gui

Please do not reply to the emails generated by GitHub issues.  You should add your update directly to the issue so that we have a permanent record of the issue and its resolution.

Is your issue now resolved?  If so, please close the issue - thanks!

prachi789, 2021-02-05:
Hi Mike,

I've run the test (cv32/tests/program/custom/interrupt_test) and passed this test in UVM environment.
I've attached two images in that you can see that only 4 values are coming in instr_addr (program counter value) signal ( 0,4,8,C) and again it's start from 0. This is happen through out the simulation.

But in my objdump file you can see that mie register is updated at 364.
I've check this register but is showing me default value.

[image]
[image]
[image]

Thanks in Advance !

MikeOpenHWGroup, 2021-02-22:
Hi @prachi789.  I am going to close this issue now.  I have not had any clear information about the original issue's current status for 18 days.   Also, in the future, please do not introduce a new issue into an existing one.  If there is a new bug, it should have a new issue.

Lastly, in the future, please provide the exact command lines to allow me (or another member of the OpenHW community) to reproduce the issue.  short descriptions and snipits of waves are rarely sufficient.

closed 2021-02-22 by MikeOpenHWGroup