openhwfoundation/core-v-verif #457: Issue in hex file dump process

prachi612, 2021-02-10:
Hi,

Right now I'm working on RISC-V UVM Environment in that i'm facing issue in hex file dump process.

In my simulation waveform  value of instr_rdata signal in ram_i module is x (don't care) throughout the simulation.

[image]

Thanks in advance,

MikeOpenHWGroup, 2021-02-10:
Hi @prachi612, thanks for creating this issue.  I will need a _lot_ more information to be able to help you.   When you created this issue there was a template that asked for things such as the steps to reproduce.  I would be particularly interested in the command-lines you used to clone the repo and run this test.  For example, I will typically do this:
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32/sim/uvmt_cv32
$ make test SIMULATOR=xrun TEST=hello-world WAVES=1
```
From the look of the waveform it seems that the test-program was not compiled or loaded properly.  In the logfile you should see a successful compile of the test-program.  For example:
```
*******************************************************************************************
* Compiling BSP
*******************************************************************************************
make -C /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/bsp RISCV=/opt/riscv RISCV_PREFIX=riscv32-unknown-elf- RISCV_EXE_PREFIX=/opt/riscv/bin/riscv32-unknown-elf-
make[3]: Entering directory '/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/bsp'
/opt/riscv/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic -c crt0.S -o crt0.o
/opt/riscv/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic -c handlers.S -o handlers.o
/opt/riscv/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic -c syscalls.c -o syscalls.o
/opt/riscv/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic -c vectors.S -o vectors.o
/opt/riscv/bin/riscv32-unknown-elf-ar rcs libcv-verif.a crt0.o handlers.o syscalls.o vectors.o
make[3]: Leaving directory '/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/bsp'
make[2]: Leaving directory '/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/sim/uvmt_cv32'
*******************************************************************************************
* Compiling test-program /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.c
*******************************************************************************************
/opt/riscv/bin/riscv32-unknown-elf-gcc -DNO_PULP \
	 \
	-Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic \
	-o /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.elf \
	-nostartfiles \
	/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.c \
	-T /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/bsp/link.ld \
	-L /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/bsp \
	-lcv-verif
*******************************************************************************************
* Generating hexfile, readelf and objdump files
*******************************************************************************************
/opt/riscv/bin/riscv32-unknown-elf-objcopy -O verilog \
	/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.elf \
	/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.hex \
	--change-section-address  .debugger=0x3FC000 \
	--change-section-address  .debugger_exception=0x3FC800
/opt/riscv/bin/riscv32-unknown-elf-objcopy: --change-section-vma .debugger_exception=0x00000000003fc800 never used
/opt/riscv/bin/riscv32-unknown-elf-objcopy: --change-section-lma .debugger_exception=0x00000000003fc800 never used
/opt/riscv/bin/riscv32-unknown-elf-objcopy: --change-section-vma .debugger=0x00000000003fc000 never used
/opt/riscv/bin/riscv32-unknown-elf-objcopy: --change-section-lma .debugger=0x00000000003fc000 never used

/opt/riscv/bin/riscv32-unknown-elf-readelf -a /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.elf > /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.readelf

/opt/riscv/bin/riscv32-unknown-elf-objdump -D -S /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.elf > /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.objdump
```

If you see any errors during this phase, the hexfile required by the simulation will not be produced.  If that is the case, the Makefiles will exit, so it is not clear to me how your simulation was launched.  Once simulation starts, the logfile will provide an indication about whether or not the hexfile was loaded properly.
```
UVM_INFO @ 16.800 ns : uvmt_cv32_dut_wrap.sv(120) reporter [DUT_WRAP] loading firmware /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tests/programs/custom/hello-world/hello-world.hex
```

prachi612, 2021-02-11:
Hi,

I'm following same step but still I'm getting instr_rdata value  is x (don't care).
In below snapshot you can see that it prints the one `uvm_info. In that  it prints one statement like (loading .hex file) but still I'm not getting proper value at instr_rdata  ram port.
[image]

[image]

Thanks in advance.

MikeOpenHWGroup, 2021-02-11:
Hi @prachi612.  I do see that you are not using the correct command-line.  It _should_ be:
```
make test SIMULATOR=vsim TEST=hello-world WAVES=1
```
Please review the [README in the sim/uvmt directory](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/uvmt_cv32/README.md) carefully.

If that does not resolve the problem, please include more information in your report.   Attach the complete compile and run-time logs and your wavedump.   In the meantime, you could look in more detail at the [instruction fetch bus](https://cv32e40p.readthedocs.io/en/latest/instruction_fetch/#instruction-fetch) to see if you can determine the issue.

prachi612, 2021-02-11:
Hi Mike,

It's solve.
Thanks for guiding me.

closed 2021-02-11 by prachi612