openhwfoundation/core-v-verif #461: Please test core_v_verif_mutli branch using Riviera-PRO

MikeOpenHWGroup, 2021-02-11:
Hi @dawidzim.  You may be aware that the cv32e40p, cv32e40x, cv32e40s and cva6 CORE-V cores may all be in active development at the same time.  In order to support multiple simultaneous core verification projects, a major update to the structure of the core-v-verif repository is underway.  This update is now available on the `core_v_verif_mutli` branch and has been extensively tested using Questa, Xcelium and Dsim.   It would be helpful to have it tested using Aldec Riviera-PRO before merging it the master branch.

The overall structure has not changed significantly, but you will notice that the `cv32` directory is renamed to `cv32e40p` and some of the `_cv32` post-fixes have been removed.  The READMEs have been updated to describe the new directory and file structure.  There are some very minor changes to the usage, but its all well documented and mostly the same as before.

Running the hello-world test is almost the same as before.  Please try the following:
```
$ git clone -b core_v_verif_multi https://github.com/openhwgroup/core-v-verif core_v_verif_multi
$ cd core_v_verif_multi/cv32e40p/sim/uvmt
$ make test SIMULATOR=riviera TEST=hello-world
```

If that works, try to run a riscv-dv generated test:
```
$ make corev-dv CV_CORE=cv32e40p SIMULATOR=riviera
$ make gen_corev-dv test TEST=corev_rand_interrupt SIMULATOR=riviera
```

Please let me know via this issue if you have any trouble.  Thanks for your help!

dawidzim, 2021-02-15:
Hey @MikeOpenHWGroup !
PR #466 resolve this issue

dawidzim, 2021-02-24:
Hi @MikeOpenHWGroup 
In #480 I restore core tests for Riviera-PRO, I think after merge in, we can close this issue

MikeOpenHWGroup, 2021-02-24:
Terrific, thanks @dawidzim

closed 2021-02-24 by MikeOpenHWGroup