openhwfoundation/core-v-verif #462: Please test core_v_verif_mutli branch using Synopsys VCS

MikeOpenHWGroup, 2021-02-11:
Hi @masgia .  You may be aware that the cv32e40p, cv32e40x, cv32e40s and cva6 CORE-V cores may all be in active development at the same time.  In order to support multiple simultaneous core verification projects, a major update to the structure of the core-v-verif repository is underway.  This update is now available on the `core_v_verif_mutli` branch and has been extensively tested using Questa, Xcelium and Dsim.   It would be helpful to have it tested using Synopsys VCS before merging it the master branch.

The overall structure has not changed significantly, but you will notice that the `cv32` directory is renamed to `cv32e40p` and some of the `_cv32` post-fixes have been removed.  The READMEs have been updated to describe the new directory and file structure.  There are some very minor changes to the usage, but its all well documented and mostly the same as before.

Running the hello-world test is almost the same as before.  Please try the following:
```
$ git clone -b core_v_verif_multi https://github.com/openhwgroup/core-v-verif core_v_verif_multi
$ cd core_v_verif_multi/cv32e40p/sim/uvmt
$ make test SIMULATOR=vcs TEST=hello-world
```

If that works, try to run a riscv-dv generated test:
```
$ make corev-dv CV_CORE=cv32e40p SIMULATOR=vcs
$ make gen_corev-dv test TEST=corev_rand_interrupt SIMULATOR=vcs
```

Please let me know via this issue if you have any trouble.  Thanks for your help!

masgia, 2021-02-18:
Hi Mike,

I've been quite busy lately, sorry for the delay in the answer.
Sure, I'll do that - most likely at the beginning of next week.

MikeOpenHWGroup, 2021-02-18:
No problem Massimiliano.  FYI, as of #469, these changes are now on the main (master) branch, so the "instructions to reproduce" are now:
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core_v_verif/cv32e40p/sim/uvmt
$ make test SIMULATOR=vcs TEST=hello-world
```

Once you have that and the `corev-dv` test (above) working, you should also be able to manually run the CI regression:
```
$ cd core_v_verif/bin
$ ./ci_check -s vcs --core cv32e40p
```

I expect (hope?) that the only changes you'll need to make are to the VCS-specific Makefile at `mk/uvmt/vcs.mk`.  If you have any questions about the updated structure of the Makefiles, they are _should_ be answered in the [README](https://github.com/openhwgroup/core-v-verif/blob/master/mk/uvmt/README.md) at `mk/uvmt`.

Yes, it looks like a lot, but its primarily structural, not functional.

masgia, 2021-02-19:
See [PR#473](https://github.com/openhwgroup/core-v-verif/pull/473)

MikeOpenHWGroup, 2021-02-19:
Thanks!   Closing this issue.

closed 2021-02-19 by MikeOpenHWGroup