openhwfoundation/core-v-verif #472: Issue in exception_pc signal

prachi612, 2021-02-19:
Hi,

I'm running interrupt testcase(cv32/tests/programs/custom) using Makefile (cv32/sim/uvmt_cv32).
You can see in below diagram value of irq_id signal is 11 ( which is indicates Machine external interrupt)  but In next cycle exception_pc signal which holds address of ( m_fast1_irq_handler).
but actually it should holds address of (m_external_irq_handler).

[image]
[image]

Thanks in Advance.

MikeOpenHWGroup, 2021-02-19:
@prachi612, when filing an issue, please follow the issue template provided.  There is specific information that you must provide in order for anyone to reproduce your issue.  At a minimum you should include the _exact_ command-lines used that brought the issue to you attention, starting from `git clone` and ending with `make test`.

For your specific issue, I see no problem.  There is no requirement that the value held by an internal signal hold a value that you expect.  The value of `exception_pc`is only important when it is used to calculate the PC that is used to fetch the next instruction.  If the value of PC is correct, that is what matters.

prachi612, 2021-02-20:
Hi Mike,

In above diagram you can see that  irq_id signal is 11 which indicates (machine external interrupt) but it select <m_fast2_irq_handler> instead of <m_external_irq_handler>.

MikeOpenHWGroup, 2021-02-20:
@prachi612, I do not think you actually read my comment.  I clearly said:

> When filing an issue, please follow the issue template provided. There is specific information that you must provide in order for anyone to reproduce your issue. At a minimum you should include the [...]

Unless I can reproduce your results, I cannot help you.  A waveform snip is not sufficient.

Silabs-ArjanB, 2021-02-22:
Hi @prachi612 

> You can see in below diagram value of irq_id signal is 11 ( which is indicates Machine external interrupt) but In next cycle exception_pc signal which holds address of ( m_fast1_irq_handler).
> but actually it should holds address of (m_external_irq_handler).

The irq_id signal you show is in hexadecimal 11, 0x11, so decimal 17 which corresponds to m_fast1_irq_handler, see https://cv32e40p.readthedocs.io/en/latest/control_status_registers/#machine-interrupt-pending-register-mip

prachi612, 2021-02-22:
Hi @Silabs-ArjanB 

This issue has been fixed.

Thanks for guiding me.

closed 2021-02-22 by prachi612