openhwfoundation/core-v-verif #483: add configuration to the results directory path

strichmo, 2021-02-26:
### Task Outcome
The common and simulator makefiles should add the configuration in the directory for all output products.

Previously, the simulation results directory would use these locations:
```
cv32e40p/sim/uvmt/<SIMULATOR>_results/xcelium.log
cv32e40p/sim/uvmt/<SIMULATOR>_results/hello-world/xrun-hello-world.log
cv32e40p/sim/uvmt/<SIMULATOR>_results/hello-world/trace_core_00000000.log
```

After this change a new directory pertaining to the configuration should be included in the path.
cv32e40p/sim/uvmt/<SIMULATOR>_results/<CFG>/xcelium.log
cv32e40p/sim/uvmt/<SIMULATOR>_results/<CFG>/hello-world/xrun-hello-world.log
cv32e40p/sim/uvmt/<SIMULATOR>_results/<CFG>/hello-world/trace_core_00000000.log

### Background information
All supported configurations are in the following directory as YAML files:
```
cv32/tests/cfg/
```

### Location Information
```
core-v-verif/mk
```

### Completion Criteria
- ci_check tests regressions should work properly with different configurations across all simulators.
- Simultaneous tests with different configurations should be executable.

MikeOpenHWGroup, 2021-03-04:
Hi @strichmo and @silabs-hfegran.  I should have picked up on this earlier:

1. The "Task Outcome" appears to have a markdown typo.  For example, I believe `cv32e40p/sim/uvmt/_results//xcelium.log` is intended to be `cv32e40p/sim/uvmt/<simulator>_results/<cfg>/xcelium.log`.  Please confirm.
2. We should also update the appropriate README to explain the use of the `CFG` variable, whether or not it has a default value and the impact of _not_ setting it.

strichmo, 2021-03-24:
Was merged and reviewed in E status meeting.

closed 2021-03-24 by strichmo