openhwfoundation/core-v-verif #488: next-gen ISA coverage model: create uvma_covg and integrate with OBI monitor

strichmo, 2021-03-01:
### Task Outcome
A UVM agent is developed to track functional coverage.  This should be an extensible, reusable UVM agent that be reused against all cores.

- Input of coverage model comes from an analysis port.  This will initially be OBI for the purpose of this task, but will be extended to RVFI in future releases.
- Have no connections to the ISS.
- Logs all instructions using standard UVM template monitor logging capabilities.
- Must only track ISA-specified instructions.  No pseudo-instructions.
- Enables dynamic configuration via configuration class of supported extensions
- Initially supports CV32E40P (RV32IMCZifencei) with ability to extend beyond.
- Creates classes based on standard UVM object conventions to model instructions

### Background information

### Location Information
core-v-verif/uvm_agents/uvma_covg

### Completion Criteria
The UVM agent is released and demonstrated on the CV32E40P meeting all above requirements.

### Additional context
Conceptual block diagram: credit @MikeOpenHWGroup 
[OBI_Coverage.docx](https://github.com/openhwgroup/core-v-verif/files/6062590/OBI_Coverage.docx)

strichmo, 2021-03-24:
Notes on implementation.  @silabs-robin has started to implement the Agent and has some basic infrastructure together.  https://github.com/silabs-robin/core-v-verif/tree/isa_agent_newcg_repeat

Based on that work I would like to put together some more detailed requirements

## Assumptions
- All of the following is contained in a uvm_agent _uvma_isa_covg_ 
- HINTs are not included

## Enumerated types
- Create an enumerated type for ISA instruction opcodes.  Currently include the IMC types.  Each type name should match ISA and be organized by RISCV extension.  No pseduo-ops.
- Create an enumerated type for ISA instruction _group_.  The types represent the general operation that the processor performs and groups instructions together.  (ALU, LOAD, STORE, BRANCH, JUMP, CSR, MMU, EXCP, PRIV, WFI).  
Each instruction opcode belongs to one and only one group.
Note that the term should be _instruction group_ for this enum.  Instruction type refers to the ISA instruction type (i.e. R-type, I-type, S-type, etc.)
Create a package level function or static method in instr class to fetch group from instruction)
- Enumerated type for GPR registers.  Name each register via ISA name (x<n>) and not the ABI name.   The ABI name should never show up in the coverage model.
- Enumerated type for CSR names
- Enumerated type for modes

## Instruction class
- Create a full UVM class to encapsulate all data fields possible within an instruction: (uvma_isa_covg_instr.sv)
- Includes all possible fields (example)
  opcode enum
  bit [4:0] rs1; // Should be enumerated
  bit [4:0] rs2;
  bit [4:0] rd;
  bit [11:0] immi;
  bit [11:0] imms;
  bit [12:1] immb;
  bit [19:0] immu;
  bit [20:1] immj;
  bit [11:1] c_immj;

- Include instruction status bits that should be extractable from RVFI
| Branch taken, not taken
| Register values
| CSR value
| Interrupt (first instruction of interrupt)
| Debug halt (first instruction of debug handler)
| Exception (first instruction of exception)

## Configuration class

There will be a configuration class to control what coverage is dumped and also configure the coverage points collected.

The intention here is to be rather aggressive with excluding coverage in the code-based versus relying on post-processing<br> exceptions.  Not only is this extra maintenance, it's also extra maintenance that is completely different from simulator-to- simulator.  The ability to _configure_ coverpoints in code is quite limited due to restrictions in SystemVerilog itself.

- enabled = Disables entire coverage model if not set
- ext_(imc)_enabled = Disables any combination of extensions, one bit per extension
- ext_zicsr_enabled = Same as above
- (mup)_mode_enabled = Removes or adds mode from instruction opcode cross with processor mode
- reg_cross_enabled = When set all instructions will add a cross of rs1,rs2,rd (for legal values)
- instr_seq_x2_enabled = When set crosses of sequential instructions of length 2 are enabled
- instr_seq_x3_enabled = When set crosses of sequential instructions of length 3 are enabled
- instr_group_seq_x2_enabled = When set cross of sequential instructions of length 2 are enabled
- instr_group_seq_x2_enabled = When set cross of sequential instructions of length 3 are enabled 

## Instruction covergroups

### Instruction type covergroup 

Each instruction type has its own coverage group.

```

```

silabs-robin, 2021-03-25:
Hi @strichmo . Should all of this go in the same task/PR?
Maybe it should be split up? My branch already has >1000 lines added.

silabs-robin, 2021-04-09:
One more thing regarding "requirements":

> it would be nice at some point to have deliberate functional coverage for specific classes of illegal instructions

As stated in https://github.com/openhwgroup/core-v-verif/pull/554#discussion_r609639523

Silabs-ArjanB, 2021-04-23:
It would be nice if the sequential instruction type cover group would also take into account whether a preceding instruction generates a result that is used by a following instruction or not (and cover all possible crosses of that) as it is that type of information that is used to decide whether hazards occur or not and whether bypasses are triggered or not.

MikeOpenHWGroup, 2021-04-23:
> It would be nice if the sequential instruction type cover group would also take into account whether a preceding instruction generates a result that is used by a following instruction or not 

Great idea @Silabs-ArjanB.  In my previously lives this has typically been done.  However, in those cases, there were known and documented rules about resource hazards (e.g. a CSR used as an rd cannot be used as an rs for two clock cycles).  AFAIK, the RISC-V ISA does not define such hazards and I would have expected formal verification to find any bugs in this area if we had them in RTL.

Nevertheless, I like the idea.  The way to get this included in our coverage model is to ensure that DV plan specifies it.  Please create an issue against the [DVplan in core-v-docs](https://github.com/openhwgroup/core-v-docs/tree/master/verif/CV32E40P/SimulationVerificationPlan).   Thanks.

silabs-robin, 2021-12-15:
AFAICT, everything here has been addressed.
@MikeOpenHWGroup and/or @silabs-hfegran , even if this issue was assigned to me I can't see that I have sufficient github-permissions to close it.

MikeOpenHWGroup, 2021-12-15:
Agreed, this issue can be closed.

@silabs-robin, as a Contributor you _should_ be able to close issues.  I will look into this.

closed 2021-12-15 by MikeOpenHWGroup

silabs-robin, 2021-12-15:
> @silabs-robin, as a Contributor you _should_ be able to close issues. I will look into this.

@MikeOpenHWGroup it is me again, Robin. Henrik is the one with the contributor rights :)

MikeOpenHWGroup, 2021-12-15:
@silabs-robin, you are partially right: @silabs-hfegran is a newly minted _Committer_, and you are a _Contributor_.  As a Contributor, you have the ability to issue pull-requests plus open and close issues.   In actual fact, _anyone_ can issue a pull-request, but OpenHW can only accept pull-requests from someone who is a Contributor (that is, covered by a Contributor Agreement).

Committers such as Henrik, can approve and merge pull-requests.