openhwfoundation/core-v-verif #524: implement rvfi rvvi scoreboard

strichmo, 2021-03-18:
### Task Outcome
Using the cv32e40p testbench reimplement step and compare using a scoreboard.  The tracer should be replaced by an RVFI interface from the processor.  The 

### Background information
The RVFI is documented here: https://github.com/SymbioticEDA/riscv-formal/blob/master/docs/rvfi.md
The RVVI is documented here: https://github.com/openhwgroup/core-v-docs/blob/master/verif/MeetingPresentations/Verif-TG-01.21.2021-RVVI.pdf

### Location Information
Reusable, configurable UVM agents will be developed for RVFI and RVVI.

### Completion Criteria
The UVM agents will be developed in adherence to established base classes.
A core-specific environment scoreboard will be developed for the cv32e40p.
PC and GPR checks will be implemented as in step and compare.
Control of the ISS will be implemented by the RVVI agent.
Interrupt and halt signaling will be re-implemented in a more reusable fashion by the RVVI agent.

strichmo, 2021-06-03:
Unassigning myself and assigning to @MikeOpenHWGroup for further assignement.  I am using the E40X for RVFI integration and do not have the time to invest in the E40P.

MikeOpenHWGroup, 2021-06-03:
Thanks @strichmo.

Hi @davideschiavone, can you provide an update on your efforts to integrated RVFI into CV32E40P?

---mike

MikeOpenHWGroup, 2023-12-20:
This issue/task is out dated and can be closed:
- the RVFI-Tracer was fully implemented on the CV32E40P v2.0
- the CV32E40S and CV32E40X have complete [RVFI-Tracer](https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/rvfi.html) implementations.
- The scoreboarding component has been implemented and documented [here](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/cv32_env.html#step-and-compare-2-0).

Note that the scoreboard is optional for environments using ImperasDV.

This issue can be closed as complete.

closed 2023-12-20 by MikeOpenHWGroup