openhwfoundation/core-v-verif #549: Add runtime control for RM to `ch_check`

MikeOpenHWGroup, 2021-04-06:
Hi @strictmo.  In #548 you added run-time control of the ISS Reference Model (which works fine) and alluded to a command-line argument for `bin/ci_check` that would disable the RM for ci-check regressions.   AFAICT that switch is not "in" and the only way to control whether the ISS is used as the RM is via the `.metrics.json` control file.

I would be good to have both:
- command-line argument to add run-time disable of all sims in a ci_check regression, regardless of control file setting
- control file control for individual tests

strichmo, 2021-04-06:
@MikeOpenHWGroup 

The switch to ci_check was included in 548:

  --iss ISS             Force USE_ISS flag to each test run (use 0 or 1),
                        default: Enabled

This forces the USE_ISS switch for all make calls.

What is not yet supported is test specification YAML.  This needs a write-up proposal because I see multiple _ways_ to specify USE_ISS right now and we need a very defined order of evaluation.  I'll work on that and append to this issue.

MikeOpenHWGroup, 2021-04-06:
> The switch to ci_check was included in 548:
Ah.  #548 was a merge to the 40x/dev branch only.  (Gotta remember to pay attention to that now.)  At some point we'll want that on the master.

strichmo, 2021-05-13:
This should be on master now with the recently completed merge.

https://github.com/openhwgroup/core-v-verif/pull/586

closed 2021-05-13 by strichmo