openhwfoundation/core-v-verif #575: Design Verification plan for all CV32E cores should include cross coverage related to instruction dependencies

Silabs-ArjanB, 2021-04-26:
During CV32E40P verification cross coverage was performed for sequences of 2 instructions. Given the pipeline architecture of the CV32E cores we should really look at coverage for sequences of 3 instructions (one in ID; one in EX; on in WB).

Not only instruction group crosses should be covered but also the cross of their dependencies (i.e. does a preceding instruction generates a result that is used by a following instruction or not). It is this dependency information that is used to decide whether hazards occur or not and whether bypasses are triggered or not.

Part of this is covered in https://github.com/openhwgroup/core-v-verif/issues/488

MikeOpenHWGroup, 2021-04-26:
I like this one @Silabs-ArjanB  and agree that it is a needed enhancement, although we need to be intelligent about how its implemented.  Simply coding up cross-coverage of sequences of three instructions will yield on the order of 10**6 bins, many of which will be low value.  Similarly, creating bins for any and all resource hazards can yield huge coverage models and only a fraction of bins are valuable.   Typically there is information that can be gleaned from either a micro-architecture spec of even the RTL that helps to focus the cross-coverage.   Let's discuss the best way to handle that at the next CTG meeting.

We should absolutely have coverage on trigger bypasses.

strichmo, 2021-04-27:
Thanks for the issue @Silabs-ArjanB.  I also agree with @MikeOpenHWGroup that we need to have some intelligent binning of instruction types to keep these sequences and crosses from getting out of hand.  

Here's a proposal for instruction types: 
LOAD => {LB, LH, LW, LBU, LHU}
STORE => {SB,SB,SW}
ALU => {Shifts, ADD,ADDI,SUB,LUI,AUIPC, XORs, ANDs, ORs, Compares}
BRANCH => {BEQ, BNE, BLT, BGE, BLTU, BGEU}
JUMP => {JAL, JALR}
FENCE => {FENCE}
FENCE_I => {FENCI.I}
ENV => {ECALL, EBREAK}
RET => {*RET}
WFI => {WFI}
CSR => {CSRRW, CSRRS, CSRRC, CSRWI, CSRRSI, CSRRCI}

Compressed instructions map into above group:
LOAD => {C.LW, C.LWSP, C.LD, C.LDSP}
STORE => {C.SW,C.SWSP,C.SD,C.SDSP}
ALU => {C.SLLI, C.ADD*, C.LI, CLUI, C.MV, C.SUB}
BRANCH => {C.BEQZ, C.BNEZ}
JUMP => {C.J, C.JR, C.JAL, C.JALR}
ENV => {C.EBREAK}

Multiplication:
MUL => {MUL,MULH, MULHSU, MULHU}
DIV => {DIV, DIVU}
REM => {REM,REMU}

Atomic: 
ALOAD => {LR.W}
ASTORE => {SC.W}
AMEM => {AMOSWAP, AMOADD, AMOAND, AMOOOR, AMOXOR, AMOMAX, AMOMIN}

These groups will be used in conjunction with general fields in the coverage instruction class (rd, rs1, rs2, and valid bits for each) to construct hazard models so that we can cross instruction_type->instruction_type with and without hazards.  This will be extensible to the effective hazard pipe length (which I believe would be 3 for the E4* cores).

Please comment on the above grouping.  (Are their other groups needed?  Do these encapsulate the anticipated hazards?)

As @MikeOpenHWGroup observed this is tied into the uarch of the processor so we could make the instruction classification a core-specific grouping as opposed to existing within the general ISACOV model that is shared amongst all CV cores.

strichmo, 2021-07-01:
This concept is now integrated into the ISACOV coverage model, which is the default for the E40X core and will be propagated to all new cores.

closed 2021-07-01 by strichmo