openhwfoundation/core-v-verif #577: Assertions not enabled during simulation

silabs-oivind, 2021-04-28:
Assertions in the CV32E40X repo are guarded by ASSERT_ON in cv32e40x_wrapper.
After the restructuring of core-v-verif, simulations are run with e.g. CV32E40X_ASSERT_ON, resulting in assertions not being enabled.
We need to agree on the naming of the assert guard, and preferably go back to ASSERT_ON.

MikeOpenHWGroup, 2021-04-28:
> We need to agree on the naming of the assert guard

Agreed.

> and preferably go back to ASSERT_ON.

I'm less sure about this.  Lots of IP uses `ASSERT_ON` and because compiler macros are global, this can make assertion management painful.

Assuming that all RTL assertions have been moved out of the RTL and into the wrapper, perhaps we no longer need a guard?

strichmo, 2021-04-28:
If I put on my "general IP provider" hat, I believe that something too general like ASSERT_ON is a bit onerous for users of the core.

From a verification perspective, assertions really should be opt-out rather than opt-in.  Additionally, let's uniquify the preprocessor directive we use:

Could we move to a preprocessor directive usage such as:
`ifndef COREV_ASSERT_OFF

If we don't want that, then we can do:
`ifdef COREV_ASSERT_ON

MikeOpenHWGroup, 2021-04-28:
The `COREV_` prefix is an improvement (although I still have a strong dislike for preprocessor directives).   Of the two options, I prefer "opt-out", so `ifndef COREV_ASSERT_OFF gets my vote.

strichmo, 2021-05-18:
Confirmed assertions are included by default now and COREV_ASSERT_OFF is being used as described above.

closed 2021-05-18 by strichmo