openhwfoundation/core-v-verif #596: $display() in coverage model

MikeOpenHWGroup, 2021-05-12:
[Line 533 of uvma_isacov_cov_model.sv](https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L533) has a call to `$display()`.  These are not to be used in UVM environments.  (I _should_ have caught this during an earlier review of your pull-request.)

Please replace the `$display()` call with a instance of the `uvm_info()`.  The verbosity should be UVM_DEBUG and it should not be an error or warning for several reasons:
1. Coverage models should not make value judgement about errors, they exist merely to record what happened.
2. We have a number of testcases that deliberately issue mal-formed instructions (to exercise the instruction decoder).  In these cases, the "no match found" condition is not an error.
3. At least one test-program sends ~100K illegal instructions, so if the verbosity is too low you get a large logifle.

silabs-robin, 2021-05-21:
This was fixed as part of https://github.com/openhwgroup/core-v-verif/commit/621b642d6cf6a1edf4fe072efac38978efc3c712 .
So the issue can be closed.
I don't see a close button so it seems I don't have permissions to do it.

MikeOpenHWGroup, 2021-05-21:
Thanks @silabs-robin.

closed 2021-05-21 by MikeOpenHWGroup