openhwfoundation/core-v-verif #599: debug_test in ci_check is failing

strichmo, 2021-05-13:
### Type
Functional failure in test

### Steps to Reproduce
% makeuvmt test TEST=debug_test

Failure does not seem to be seed-specific

### Additional context
The debug_test in ci_check for the cv32e40x is failing and should be fixed as soon as possible.  The test began failing after updating the RTL pointer in the cv32e40x common.mk

UVM_ERROR @ 8776113.300 ns : uvmt_cv32e40x_step_compare.sv(117) reporter [Step-and-Compare] PC expected=0x000067ec and actual=0x1a110800 PC=0x000067ec
UVM_ERROR @ 8776113.300 ns : uvmt_cv32e40x_step_compare.sv(117) reporter [Step-and-Compare] GPR[14] expected=0x00000000 and actual=0x07000301 PC=0x000067ec
UVM_ERROR @ 8776113.300 ns : uvmt_cv32e40x_step_compare.sv(117) reporter [Step-and-Compare] dscratch0 expected=0x0000e7bc and actual=0x0000d070 PC=0x000067ec
UVM_ERROR @ 8776134.300 ns : uvmt_cv32e40x_step_compare.sv(117) reporter [Step-and-Compare] PC expected=0x000067f0 and actual=0x1a110804 PC=0x000067f0
UVM_ERROR @ 8776134.300 ns : uvmt_cv32e40x_step_compare.sv(117) reporter [Step-and-Compare] GPR[10] expected=0x0000d070 and actual=0x1a111804 PC=0x000067f0

silabs-oysteink, 2021-05-14:
https://github.com/openhwgroup/core-v-verif/pull/601 fixes this issue.

strichmo, 2021-05-14:
Merged PR.  Will close when triggered Action passes.

strichmo, 2021-05-14:
Fixed.

closed 2021-05-14 by strichmo