openhwfoundation/core-v-verif #612: RVFI UVM step and compare with interrupt and debug request

strichmo, 2021-05-20:
### Task Outcome
The RVFI/RVVI implementation completed in this following issue properly supported interrupts and debug requests.

### Completion Criteria
- The interrupt and debug request signaling is integrated into the RVVI interfaces.
- Existing E40P interrupt and debug tests pass on the E40X.  Any failures can be waived for purposes of this issue if they are related to RTL issues.

MikeOpenHWGroup, 2021-05-20:
Another one to follow @JeanRochCoulon

strichmo, 2021-07-01:
Scoreboard is running in wb_controller_e40x branch.
Old step and compare is removed and deprecated.
Temporary USE_RVVI flag is also deprecated as this mode of checking is now default.

closed 2021-07-01 by strichmo