openhwfoundation/core-v-verif #613: RVFI UVM step and compare with CSRs

strichmo, 2021-05-20:
### Task Outcome
The RVFI/RVVI testbench can check CSRs to the equivalent functionality provided in the current module-based Step and Compare.

### Completion Criteria
- The RVFI implementation in E40X should be updated to support CSRs.
- The checker scoreboard should be extended in E40X to check CSRs at retirement
- Methodology must be developed and implemented to control CSR checking for CSRs that may be unpredictable in certain instruction retirement condition.

MikeOpenHWGroup, 2021-05-20:
@JeanRochCoulon

strichmo, 2021-07-01:
Scoreboard is running in wb_controller_e40x branch.
Old step and compare is removed and deprecated.
Temporary USE_RVVI flag is also deprecated as this mode of checking is now default.

closed 2021-07-01 by strichmo