openhwfoundation/core-v-verif #614: Document RVFI/RVVI implementation in verification strategy

strichmo, 2021-05-20:
### Task Outcome
The OpenHW Verification Strategy Document  should be updated to document RVFI, RVVI and general scoreboard implementation for CORE-V cores.

### Location Information
The source for the ReadTheDocs (restructed text) live here:
https://github.com/strichmo/core-v-docs/tree/master/verif/Common/CORE-V_Verification_Strategy

But this may be moved at a future date.  This issue will be updated if so.

### Completion Criteria
The document is updated and it is verified that the rendering infrastructure works properly with the updates.

MikeOpenHWGroup, 2021-05-20:
Last one @JeanRochCoulon, I promise.  :wink:

strichmo, 2021-07-01:
Initial documentation is here:
https://core-v-docs-verif-strat.readthedocs.io/en/latest/

closed 2021-07-01 by strichmo