openhwfoundation/core-v-verif #619: DSIM incorrectly failing mip interrupt assertion

strichmo, 2021-05-21:
### Type
* Tool error

This appears to be a tool issue with the latest DSIM version against both the cv32e40p and cv32e40x.
This assertion will fire:
UVM_ERROR @ 1615.800 ns : uvmt_cv32e40x_interrupt_assert.sv(301) reporter [CV32E40X_IRQ_ASSERT] MIP of 0x00000000 does not follow flopped irq_i input: 0x00200500

It appears to be related to clock gating during WFI as the assertion should not be evaluated during that time, but root-cause analysis is on-going at Metrics.

### Steps to Reproduce
Run any corev-dv random test that introduces WFI instructions.

*Note that many corev-dv interrupt-related tests are now disabled for dsim in the rel_check yaml regression lists.  When this issue is resolved to close this issue those tests must be restored as a master-hotfix and propagated to the dev branches.*

Silabs-ArjanB, 2021-05-21:
If there is an issue in the 40P as well then you can ignore above remark.

strichmo, 2021-05-24:
@Silabs-ArjanB Yes the issue manifests in both core (e40x,e40p) so it appears systemic with DSIM as opposed to anything core-specific.

strichmo, 2021-07-01:
Fixed in an earlier DSIM release.  The release has been propagated to all cores.

closed 2021-07-01 by strichmo