openhwfoundation/core-v-verif #632: vplan review for PMA

strichmo, 2021-05-25:
### Task Outcome
The verification plan for PMA functionality is reviewed for CV32E40X.  This plan will be adapted to the CV32E40S later, but that plan will be refined and reviewed at a later time.

### Background information
The current PMA vplan is found here: https://github.com/openhwgroup/core-v-verif/blob/cv32e40x/dev/cv32e40x/docs/VerifPlans/Simulation/privileged_spec/CV32E40X_PMA_VerifPlan.xlsx

There will be a meeting to review the PMA vPlan on Friday 5/28 at 8am US Central.
The meeting call is here: 
Topic: CV32E40X PMA/Bus Error Vplan Review
Time: May 28, 2021 08:00 AM Central Time (US and Canada)

Join Zoom Meeting
https://silabs.zoom.us/j/94866993802?pwd=OVI0bnB4Ky96MUsrd0dya0cxUU9JQT09

**Please preview the plan beforehand and provide comments to ensure the meeting is as efficient as possible**

Provide review by either:
- Adding comments to the Excel and attaching to a comment on this Issue.
- Directly replying to this Issue

### Completion Criteria
The feedback from the review will be integrated to the vplan and added to the cv32e40x repo in the cv32e40x/dev branch via a PR.

Silabs-ArjanB, 2021-05-26:
Please find my feedback attached:
[CV32E40X_PMA_VerifPlan.xlsx](https://github.com/openhwgroup/core-v-verif/files/6544431/CV32E40X_PMA_VerifPlan.xlsx)

Silabs-ArjanB, 2021-05-26:
Having also read the bus error vplan I see that a different (more encompassing) strategy is followed there.

Please add the following to the PMA vplan as well:

- Interaction with Debug mode and single stepping
- Interaction with Zce table jumps
- Suppression of *_req_o in case of faulting memory access attempts

Silabs-ArjanB, 2021-05-26:
Hi @silabs-hfegran The write buffer is not 100% sure yet. Anyway, I will add documentation issues for both topics for the 40x.

strichmo, 2021-05-27:
@silabs-hfegran Since the PMA is parameterized, this is problematic to achieve coverage given available randomization techniques.   Any known techniques to "randomize" parameters involve a lot of infrastructure and generally are not worth the complexity they introduce.  

Therefore I would like to introduce a few fixed configurations to verify the PMA configurations.  Obviously these would need to be selected properly to ensure we verify all features.

- No regions
- Single memory region (unmapped == I/O)
- Full 16-region mapping w/overlapping regions
- 1 or 2 configurations in between

strichmo, 2021-05-27:
Would like to see some functional coverage (might work with cover props as well) for:
- Instruction group type from ISACOV with PMA instruction fetch violation
- PMA region index matched (or default I/O unmapped region) crossed with:
  - Instruction fetch (passing)
  - Load (passing)
  - Store (passing)
  - Atomic (passing)
  - Load alignment error
  - Store alignment error
  - Atomic load-reserved unallowed error
  - Atomic store-conditional unallowed error

MikeOpenHWGroup, 2021-05-27:
Hi @silabs-hfegran, good job on the PMA DV plan.  My specific comments are in the attached spreadsheet.
[CV32E40X_PMA_VerifPlan(mt-review).xlsx](https://github.com/openhwgroup/core-v-verif/files/6554783/CV32E40X_PMA_VerifPlan.mt-review.xlsx)

I also have some comments that are not specific to this DV plan and I will post them on the [TWG: Verification](https://mattermost.openhwgroup.org/all-users/channels/twg--verification) on MatterMost.

silabs-hfegran, 2021-05-28:
Thank you all for your comments and feedback! 
I'll digest these comments and modify the plan to reflect these proposed changes.

@strichmo :
I guess I either didn't use the spreadsheet correctly or didn't express myself clearly enough - for the configurations, my intention was that all the other tests in the spreadsheet should be executed with a minimum set of configurations. My intentions were as follows:

- No regions
- 1 Region (min)
- 2 regions (min + 1) 
- 15 regions (max - 1)
- 16 regions (max)
- 1 configuration with between 3 and 14 regions defined (did not specify -  something like 7 would suffice) 

If this is too many configurations to be feasible, i would skip 2 and 15 regions, which maches what you suggested. 

@MikeOpenHWGroup : 
User manual states that atomic extensions is supported in the X-core, thus included in the test plan.

closed 2021-09-16 by strichmo