openhwfoundation/core-v-verif #633: vplan review for Bus Errors

strichmo, 2021-05-25:
### Task Outcome
The verification plan for OBI bus error functionality is reviewed for CV32E40X. This plan will be adapted to the CV32E40S later, but that plan will be refined and reviewed at a later time.

### Background information

The current PMA vplan is found here: https://github.com/openhwgroup/core-v-verif/blob/cv32e40x/dev/cv32e40x/docs/VerifPlans/Simulation/privileged_spec/CV32E40X_PMA_VerifPlan.xlsx

There will be a meeting to review the PMA vPlan on Friday 5/28 at 8am US Central.
The meeting call is here:
Topic: CV32E40X PMA/Bus Error Vplan Review
Time: May 28, 2021 08:00 AM Central Time (US and Canada)

Join Zoom Meeting
https://silabs.zoom.us/j/94866993802?pwd=OVI0bnB4Ky96MUsrd0dya0cxUU9JQT09

**Please preview the plan beforehand and provide comments to ensure the meeting is as efficient as possible**

Provide review by either:
- Adding comments to the Excel and attaching to a comment on this Issue.
- Directly replying to this Issue

### Location Information
Specify the path, relative to the root of this GutHub project, of the code that
will be created and/or modified.

### Completion Criteria
The feedback from the review will be integrated to the vplan and added to the cv32e40x repo in the cv32e40x/dev branch via a PR.

Silabs-ArjanB, 2021-05-26:
Here is my feedback on the vplan: 
[CV32E40X_VerifPlan_Bus_Error.xlsx](https://github.com/openhwgroup/core-v-verif/files/6544695/CV32E40X_VerifPlan_Bus_Error.xlsx)

MikeOpenHWGroup, 2021-05-27:
My comments:
[CV32E40X_VerifPlan_Bus_Error(mt-review).xlsx](https://github.com/openhwgroup/core-v-verif/files/6555888/CV32E40X_VerifPlan_Bus_Error.mt-review.xlsx)

strichmo, 2021-05-28:
My feedback:
- Interrupt priority is not required to be covered here.  The interrupt vplan does have (and will continue to have) priority checking of the arbitration as assertions.
- There are some other comments on the latency between NMI assertion and load/store bus errors.  I would like to see a max latency on this.
- Functional coverage on instruction group vs. bus error (on instruction fetch)

strichmo, 2021-07-01:
Released to cv32e40x.

closed 2021-07-01 by strichmo