openhwfoundation/core-v-verif #638: wrong number of arguments specified for 'interrupt' attribute -- Getting this error while running interrupt test.

terracode01, 2021-06-01:
../../tests/programs/custom/interrupt_test/interrupt_test.h:69:1: error: wrong number of arguments specified for 'interrupt' attribute
 __attribute__((interrupt ("machine"))) void m_software_irq_handler(void);
 ^~~~~~~~~~~~~
../../tests/programs/custom/interrupt_test/interrupt_test.h:70:1: error: wrong number of arguments specified for 'interrupt' attribute
 __attribute__((interrupt ("machine"))) void m_timer_irq_handler(void);
 ^~~~~~~~~~~~~
../../tests/programs/custom/interrupt_test/interrupt_test.h:71:1: error: wrong number of arguments specified for 'interrupt' attribute
 __attribute__((interrupt ("machine"))) void m_external_irq_handler(void);

Getting above error while running using command -- make custom CUSTOM_PROG=interrupt_test

MikeOpenHWGroup, 2021-06-01:
Hi @terracode01, thanks for taking the time to create this issue.   Its difficult to fully diagnose your issue without additional details but it looks to me like a toolchain issue.  Information on the recommended toolchain can be found [here](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/sim/TOOLCHAIN.md).

I also notice that you are using the `make custom` target.  This target has been depreciated in favour of the `make test` target.  Please see the [Common Makefiles for Core-V UVM Verification Environment](https://github.com/openhwgroup/core-v-verif/blob/master/mk/uvmt/README.md) README.

Let me know if the above information helps you out.  If not, please provide the **exact** command line you used.  For example, to run the interrupt test on the CV32E40P I would do this:
```
$ git clone https://github.com/openhwgroup/core-v-verif.git
$ cd core-v-verif/cv32e40p/sim/uvmt
$ make test TEST=interrupt_test
```
(Note that the above assumes that I have set SIMULATOR and COREV variables as described in the above README.)

If you can, it would be helpful to attach the compile logs as well.

terracode01, 2021-06-02:
Hi Mike, 

What I understood was there are two tool chains COREV and PULP. Earlier I was using PULP tool chain. After updating my path to CORE V its running. 

I am using the following steps to run 

work_pulp/core-v-verif/cv32e40p/sim/core --> Running test in this DIR 

Command used to run

closed 2021-06-02 by terracode01

MikeOpenHWGroup, 2021-06-02:
Hi Rahul.   You are running the `core` testbench which does not support interrupt testing.   The `core` testbench exists as a demonstration vehicle only - if you want to fully verify the CV32E40P (or any other CORE-V cores) using core-v-verif you will need to use the UVM environment.  Note that this requires a commerical SystemVerilog simulator and in the case of CV32E40P and CV32E40X, an Imperas OVPsim license.