openhwfoundation/core-v-verif #660: Assign to member of enum in cv32e40p mm_ram.sv

dawidzim, 2021-06-07:
### Bug name
Assign to member of enum in `https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/tb/core/mm_ram.sv`

### Type
* Compile error (hopefully nobody has committed anything that doesn't compile!)

Hey!
In `mm_ram.sv` we have this fragment of code:
```
            uvmt_cv32e40p_tb.iss_wrap.ram.TICKS = data_rdata_mux;
```
TICKS is member of enum so should be treated like a constant. 
LRM 1800-2017 Chapter 6.19:
```
An enumerated type declares a set of integral named constants.
```

MikeOpenHWGroup, 2021-06-18:
Hi @dawidzim, thanks for the issue and apologies for not responding soon (this one fell through the cracks).

I agree with your interpretation of the LRM, but apparently the other simulators are more forgiving.  :-(   Presumably you are using Aldec Riviera-PRO.

Hi @silabs-mateilga, please have a look at this.  We need TICKS to be a variable, not a constant.

silabs-mateilga, 2021-06-21:
@MikeOpenHWGroup I think there might be some simulator issue here, as it seems to be confusing the mm_ram internal enum TICKS with the register TICKS in the imperas ram module.

https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/vendor_lib/imperas/design/ram.sv#L39

This is what is being assigned, but somehow the simulator used by @dawidzim seems to be mixing these up. For more details here, maybe @eroom1966 has some input? This relates to the code he added when making ISS aware of the peripheral "TICKS" cycle counter.

dawidzim, 2021-06-21:
Hey @silabs-mateilga 
thanks for pointing this, so seems like this is my mistake, so we can close this issue.
I will try to find some work around for this problem and open pull request

MikeOpenHWGroup, 2021-06-21:
Thanks @silabs-mateilga.  Closing the issue.

dawidzim, 2021-06-23:
Hey!
Problem is in Makefile, we should add files from `/cv32e40p/tb/uvmt` to compilation. Compilation command for VCS after `make sanity`, for Riviera we have the same files:
```
vcs +vc -sverilog -race=all -ignore unique_checks -full64 \
        -timescale=1ns/1ps \
        -assert svaext \
        -CC "-I/opt/synopsys/vcs-mx/O-2018.09-SP1-1/include -O3 -march=native"  \
        -f /core-v-verif/cv32e40p/sim/core//core-v-verif/core-v-cores/cv32e40p/cv32e40p_manifest.flist \
        /core-v-verif/cv32e40p/tb/core/tb_riscv/include/perturbation_defines.sv \
	/core-v-verif/cv32e40p/tb/core/tb_top.sv /core-v-verif/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv \
	/core-v-verif/cv32e40p/tb/core/mm_ram.sv /core-v-verif/cv32e40p/tb/core/dp_ram.sv \
	/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_random_stall.sv /core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_random_interrupt_generator.sv \
	/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_rvalid_stall.sv /core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_gnt_stall.sv
```
I think all simulators should report error (except Verilator because `ifndef Verilator)
@MikeOpenHWGroup @silabs-mateilga Should I open another issue for this?

MikeOpenHWGroup, 2021-06-23:
Nice catch.  I'll reopen this issue and investigate off-line.

MikeOpenHWGroup, 2021-06-24:
OK, I _finally_ understand.   The problem @dawidzim is seeing is with the `core` testbench, not the `uvm` testbench.  This issue has crept in because we have not been regressing the core testbench with simulators other than Verilator.

@dawidzim can you confirm that the above is correct?  Specifically:
1. Can you run Verilator on the core testbench?
```
$ cd core-v-verif/cv32e40p/sim/core
$ make
```
2. Does the UVM environment work for you:
```
$ cd core-v-verif/cv32e40p/sim/uvmt
$ make sanity SIMULATOR=riviera
```

dawidzim, 2021-06-25:
Sorry, I should tell about `core` testbench on the start.

@1 I have problem with compile new version of Verilator... building process takes all RAM, I will leave it overnight, and report results on monday
@2 Yes, Riviera works with UVM, also ci_check reports about PASS

btw. @MikeOpenHWGroup I send you e-mail few days ago. Can you check this? Thank you in advance!

dawidzim, 2021-07-20:
After #721 tests from core pass

closed 2021-07-20 by dawidzim