openhwfoundation/core-v-verif #662: Coverage of test cv32e40p_csr_access_test, for exception behavior

eroom1966, 2021-06-08:
Can I ask, for the test above cv32e40p_csr_access_test - does this test also ensure that CSR Registers which should NOT exist, raise an appropriate illegal instruction exception ?

so as a simple example, a Read from uip/csr[0x044] is expected to cause an illegal instruction exception

thx
Lee

MikeOpenHWGroup, 2021-06-08:
Hi @eroom1966.  The `cv32e40p_csr_access_test` makes no attempt to access non-existant CSRs.  This is a generated test based on the description at `core-v-verif/lib/corev-dv/cv32e40p_csr_template.yaml`.

The test-program that does access non-existant CSRs is `cv32e40p_readonly_csr_access_test.S`.  Rather than attempt to access every non-existant CSR address, this program attempts to access PULP CSRs which are not implemented in the CV32E40P.

eroom1966, 2021-06-08:
Hi mike
i tried to run both of these in USE_ISS mode, but was unsuccessful.
How do I make it run ?
Thx
Lee

closed 2023-02-21 by eroom1966