openhwfoundation/core-v-verif #668: OVPSIM handling of trap instructions versus interrupt instructions

strichmo, 2021-06-14:
### Type
This is a run-time failure in the RVFI/RVVI scoreboard.

### Steps to Reproduce
Use this branch:
https://github.com/openhwgroup/core-v-verif/tree/rvfi_rvvi_initial_e40x
Commit: 
7e3fb1c4299a18c813a2227ade1d95ee31144d2d

On the E40X run this test:
```
% makeuvmt test TEST=illegal WAVES=1 CV_CORE=cv32e40x USE_RVVI=1
```

The test should fail as such:
```
UVM_ERROR @ 21300.300 ns : uvme_cv32e40x_core_sb.sv(249) uvm_test_top.env.core_sb [CORESB] Received RVVI out of order: 1543, exp_order, 1542
UVM_ERROR @ 21312.300 ns : uvme_cv32e40x_core_sb.sv(249) uvm_test_top.env.core_sb [CORESB] Received RVVI out of order: 1544, exp_order, 1542
```

The issue is that the RVVI is incrementing *order* for the *trap* instructions.  For this instruction *valid* is also deasserted.  This is inconsistent with interrupts that also signal *trap* and deassert *valid*.  In that case RVVI from OVPSIM does NOT increment *order*.  Note that incrementing *order* is OK, the RVFI also increment *order* such that the instructions still match up.  The issue is that an interrupt looks exactly like a trap, so the monitor/scoreboard does not know which case to "throw out" and which to compare.

One possible solution may be to use *intr* instead of *trap* to signal state on RVVI where an interrupt is taken.  That way the scoreboard can delineate this case from the trapped instruction case.

Here's some waveforms to illustrate:

This is an illegal instruction:
[image]

This is an interrupt:
[image]

eroom1966, 2021-06-15:
@strichmo 
Hi Steve, I tried this on my local branch I have been working on - I get a different failure

UVM_ERROR @ 74946.300 ns : uvme_cv32e40x_core_sb.sv(369) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 5293, csr: mstatus, rvfi = 0x00001800, rvvi = 0x00001880

this seems much later, so I am presuming I am not seeing the earlier failure, I will investigate why I get this fail
Thx
Lee

eroom1966, 2021-06-15:
FYI @MikeOpenHWGroup @strichmo @GTumbush 

I am widening this out a little bit for comments from Mike and Greg. 
The question here I think is what is the field rvfi_order incremented by ?

Now the RVFI document says this is:-

> The rvfi_order field must be set to the instruction index. No indices must be used twice and there must be no gaps. Instructions may be retired in a reordered fashion, as long as causality is preserved (register and memory write operations must be retired before the read operations that depend on them).

If indices **cannot be used twice**, then by definition this number **must** be incremented on an instruction which takes any of the following paths

- retires (rvfi_valid)
- traps (rvfi_trap)
- or halts (rvfi_halt)

In which case in the above, an illegal instruction which will be reported as rvfi_trap, should still have a unique rvfi_order number ?
if not, this does not comply with the previous statement **No indices must be used twice and there must be no gaps**

discuss ....

MikeOpenHWGroup, 2021-06-15:
> If indices cannot be used twice, then by definition this number must be incremented on an instruction which takes any of the following paths: retires (rvfi_valid), traps (rvfi_trap) or halts (rvfi_halt)

Agreed.

> an illegal instruction which will be reported as rvfi_trap, should still have a unique rvfi_order number 

That is my read of it.

eroom1966, 2021-06-21:
@strichmo 
Steve I am trying to enable tracing for this test, I usually modify 
cv32e40x/tests/cfg/default.yaml

but this seems to be having no effect, does this flow use a different method to get the options used ?

eroom1966, 2021-06-21:
I think I can see the issue here - I am not passing the CSR's across on an exception

strichmo, 2021-06-21:
@eroom1966 I would recommend doing one of the following:

1. Unset CV_CORE from your environment, and cd to either cv32e40x/sim/uvmt or cv32e40p/sim/uvmt for whichever core you want to simulate and use __make__ 
or 
2. Use makeuvmt and set CV_CORE either in the environment or on the make command line

If you want to be absolutely sure you are simulating the right core, use #1 above.  It may help to run a clean target first.

eroom1966, 2021-06-22:
urghh, stupid mistake, I had another earlier core-v-verif/bin in my searchpath
OK, Steve I think this is fixed here 
https://github.com/eroom1966/core-v-verif/tree/rvfi_rvvi_initial_e40x_shared

if I run 
makeuvmt test TEST=illegal CV_CORE=cv32e40x USE_RVVI=1

it passes

strichmo, 2021-07-01:
Fixed in wb_controller_e40x branch in the new unified Imperas OVPSim model integration.  This will be propagated to all cores once the next merge cycle occurs.

closed 2021-07-01 by strichmo