openhwfoundation/core-v-verif #681: use OBI memory agent in the CV32E40X testbench

strichmo, 2021-06-17:
### Task Outcome
Merge the UVM OBI active memory agent into CV32E40X testbench.

### Background information
The mm_ram and all of its functionality should be replaced:
- Virtual peripherals
- Memory storage for I and D OBI busses

### Completion Criteria
The mm_ram is deprecated and all passing testing continue to pass.

MikeOpenHWGroup, 2021-06-17:
Update.  @datum-dpoulin and I are actively working on the OBI memory agent. Currently the work is on the E40P, but this can be ported to E40X.

silabs-hfegran, 2021-06-22:
@MikeOpenHWGroup Do you have a rough estimate on when the work for e40p will be ready?

MikeOpenHWGroup, 2021-06-22:
Targeting end of this week.   The first release will not support PMA.

strichmo, 2021-08-06:
The UVM OBI agent is now implemented and supported with this PR.  https://github.com/openhwgroup/core-v-verif/pull/774
The old OBI agent is now deprecated and removed from the cv32e40x testbench.

This will be merged to cv32e40x/dev shortly.

closed 2021-08-06 by strichmo