openhwfoundation/core-v-verif #687: DPI Export fails using DSIM

MikeOpenHWGroup, 2021-06-18:
The latest commit to the cv32e40x/dev branch of core-v-verif [c0920f91](https://github.com/openhwgroup/core-v-verif/commit/c0920f91673a2ec6f69f469c4f0d37831b4cdeaa) compiles and runs as expected using Cadence Xcelium (xrun), but fails at run-time under Metrics DSIM.  The runtime error is:
```
=F:[DpiExport] exported tf busFetch called but DPI context is not set.
```

### Steps to Reproduce
$ git clone -b cv32e40x/dev/ https://github.com/openhwgroup/core-v-verif.git
$ cd core-v-verif
$ git clone c0920f9
$ cd cv32e40x/sim/uvmt
$ make sanity SIMULATOR=dsim

### Additional context
All tests in the June 18 11:02 `cv32e40x_ci_check_dev` MetricsCI regression failed due to this issue.  The logfiles are available [here](https://openhwgroup.metrics.ca/main-core-v-verif-openhwgroup/results/regressionRuns/3038ec8f-1e57-4288-bbed-3c460544cf51).

This commit was known to have this issue before the merge.  It was deliberately decided to merge it in to allow for continued development on the dev branch.   See #685.

An issue has been opened with Metrics, see: Metrics Zendesk issue 1052: OpenHW: Metrics DPI context fatal failure.

strichmo, 2021-06-23:
@strichmo and @aimeepsutton had a debug session on 6/22.  From enabling more debug on the ISS, we were able to see that certain signals (deferint, haltreq, etc.) were being set to {64{1'b1}} (i.e. -1 for 64-bits).  This suggests that perhaps the struct communication from SystemVerilog->DPI->ISS is somehow not correct.  

@aimeepsutton will ping Imperas to see if they can give any clues or disclose source for the DPI layer (svPull implementation)

strichmo, 2021-07-01:
This is now fixed in the latest DSIM that has been propagated to cv32e40x/dev branch and in the wb_controller_e40x branches.  This will be propagated to master when next merge cycle completes.

closed 2021-07-01 by strichmo