openhwfoundation/core-v-verif #705: remove irq_ack_o from all interrupt testbench infrastructure

strichmo, 2021-07-02:
### Task Outcome
Testbench interrupt agent functionality is maintained but any dependence on the hardware ack mechanism (irq_id_o and irq_ack_o) is removed.  The design team is planning to remove these pins.

### Completion Criteria
Virtual sequence, monitor and properties work without irq_ack_o and irq_id_o.

MikeOpenHWGroup, 2021-07-02:
Are we removing this from _all_ interrupt testbench infrastructure?   I would expect the E40P cores to continue to use `irq_id_o` and `irq_ack_o`.  Can we control use of this ack mechanism with a cfg member?

strichmo, 2021-07-12:
Yes, this needs to be configurable.  I will need to consider a more reasonable alternative between with and without as one could easily imagine a scenario where it would be difficult to make forward progress with a random test if an interrupt line is held asserted and the interrupts is constantly being retaken.

I believe the agent will need some semblance of an "ack" based on the core micro-architecture.

silabs-robin, 2022-01-04:
Fixed by the PR that fixed https://github.com/openhwgroup/cv32e40x/issues/231

Conclusion:
The signals were kept in the RTL. But they were made internal-only and renamed to not have the "_o" suffix, and a comment explains that the signals are for verification purposes.

silabs-robin, 2022-01-04:
This issue can be closed. I don't have the permissions.

MikeOpenHWGroup, 2022-01-04:
Thanks @silabs-robin

closed 2022-01-04 by MikeOpenHWGroup