openhwfoundation/core-v-verif #707: Resurecting the XPULP tests

MikeOpenHWGroup, 2021-07-02:
All XPULP test-programs for the CV32E40P are currently failing, see attached email.  
[OpenHWGroupEmail-Resurecting_the_XPULP_tests.pdf](https://github.com/openhwgroup/core-v-verif/files/6755005/OpenHWGroupEmail-Resurecting_the_XPULP_tests.pdf)

The root cause is now known and the fix is easy - this issue exists to make sure it doesn't get forgotten.

Note that this issue affect the CV32E40P only.

This issue can be closed when all XPULP test-programs _except_ `pulp_vectorial_add_sub` and `pulp_vectorial_bit_manip` are passing on the head of the master branch.

MikeOpenHWGroup, 2022-03-15:
Hi @jstraus59 
> I am wondering what the status is of the PULP tests, particularly the vectorized (now called SIMD) instructions. So far, I have found the bit manipulation, general ALU and immediate branching tests work well, but I have found [some] issues now that I am staring to try the vectorized tests.

The PULP tests currently in the repo were donated by `em-microelectronic` who use a version of CV32E40P that was not fully verified.  Specifically, em-micro used the non-standard PULP encodings.  The tests have not been looked at since late 2020.  So I would be very careful using these tests in their current form.

jstraus59, 2022-04-01:
FYI, found additional minor errors in the cv32e40p pulp_vectorial_bit_manip test while completing the Imperas modelling of the PULP_XPULP extensions:

1) Incorrect register used for comparisons
2) One expected value was incorrect

I will create a pull request with the changes for the 2 tests found to have errors.

MikeOpenHWGroup, 2023-12-20:
Hi @dd-vaibhavjain, this issue has been long forgotten, but I believe it can be closed as the XPULP tests are running successfully on CV32E40P v2.  Can you confirm this?  Can this issue be closed?

dd-vaibhavjain, 2023-12-29:
Yes all these PULP tests are successfully running and passing in all targeted configs for CV32E40Pv2.
We can close the ticket.

closed 2023-12-29 by MikeOpenHWGroup