openhwfoundation/core-v-verif #712: DSIM fails with heap messages with wb_controller_e40x branch

strichmo, 2021-07-03:
### Type
DSIM run-time failure 
### Steps to Reproduce
Please provide:
1. URL to branch that exhibits the issue.
2. Command line  (e.g. `make xrun-riscv-compliance`)
3. Logfile and/or wave-dump info (screen shots can be useful)

### Additional context
Add any other context about the problem here.

MikeOpenHWGroup, 2021-07-13:
Just so everyone is aware: the recent merge of the `wb_controller_e40x` branch to `cv32e40x/dev` (#711) means that MetricsCI will fail.  This was a deliberate move to ensure the  `cv32e40x/dev` branch can continue to make progress.

strichmo, 2021-07-26:
The issue was an assertion that was creating lots of threads in the antecedent of a concurrent property.  The property used a goto operator as it's very first clause.  This forces the simulator to spawn a thread at each clock edge, which can severely degrade resources on long simulations.

Fixed here by using repetition operators instead to force vacuous pass when condition not present.
https://github.com/openhwgroup/cv32e40x/pull/142

closed 2021-07-26 by strichmo