openhwfoundation/core-v-verif #748: define methodology for initializing memory in TB and RM outside of defined program

strichmo, 2021-07-22:
### Type
* Methodology

The issue is that currently the testbench will initialize memory in the testbench RTL (i.e. the mm_ram) and the reference model in the ISS infrastructure to the the same values.  There are some issues with this:

* It can be considered bad test design to read values that are not explicitly initialized nor implicitly initialized via the elf (i.e. the test program).
* Out-of-bounds data memory accesses should be caught as soon as possible.  The current methodology may hide a rogue pointer or GPR pointer value well beyond the initial failure point in a test.

### Steps to Reproduce
No specific test to reproduce the issue.  The methodology needs to be defined and clarified.

The methodology will be implemented in the cv32e40x first then propagated to the other cores.

This issue will serve as a discussion and initial documentation of the solution.  Once determined the verification strategy document will be updated.

MikeOpenHWGroup, 2021-07-23:
This is not directly related, but the init code needs to be aware of the [remapping](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/tb/core/mm_ram.sv#L300) done in the mm_ram.

strichmo, 2021-12-09:
Reassigning this to @MikeOpenHWGroup as I have not made progress on the issue.