openhwfoundation/core-v-verif #754: vplan review for RV32I extension

strichmo, 2021-07-27:
### Task Outcome
Updated vplan coverage plan definition for the RV32 M extension.  This vplan should be applicable to all CORE-V processors implementing the I extension (i.e. all cores)

### Background information
This review will update the coverage definition using the new ISACOV UVM coverage agent in all CORE-V testbenches.

### Location Information
The updated excel can be fetched from this branch:
https://github.com/openhwgroup/core-v-verif/tree/strichmo/pr/isa_i_m_vplan_review/docs/VerifPlans/ISA/RV32/Simulation

### Completion Criteria
Vplan meeting is completed.
All action items completed
Vplan definition updated on master branch.

silabs-hfegran, 2021-07-29:
Looks good Steve, I have a couple of small  comments

- I find the "instr_i_<instr_name>" syntax somewhat ambiguous outside of context, this could also indicate an I-type instruction. 
My suggestion would be to do something akin to rv32i_<instr_name>. While I understand why the instr_ part is included I do find it somewhat superfluous. 

- \*\_hazard coverpoints lacks cp_ prefix

- fence/ebreak cp's are suffixed _fixed, perhaps "executed/exec" or something similar might be more illustrative.

- nitpicks: typo in JAL description -> uncomditional 

(By the way, the task outcome of this issue mentions the M-extension, I guess that's just a copy paste error)

strichmo, 2021-07-29:
Vplan meeting held and action items were updated to the above fork and will be PRed.
This issue will be closed but any further feedback is welcomed.  Simply contact @strichmo or open another issue against core-v-verif.

closed 2021-07-29 by strichmo