openhwfoundation/core-v-verif #762: Investigate build_phase implementation of `uvma_core_cntrl_agent_c`

MikeOpenHWGroup, 2021-08-01:
The [uvma_core_cntrl_agent_c](https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_core_cntrl/uvma_core_cntrl_agent.sv) agent is a component of the [core-v-verif library](https://github.com/openhwgroup/core-v-verif/blob/master/lib) of reusable UVM Agents.  The cv32e40x extends this agent to create [uvma_cv32e40x_core_cntrl_agent_c](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40x/env/uvme/uvma_cv32e40x_core_cntrl_agent.sv).

DSIM runs of the cv32e40x environment experience a noticeable, unexplained, pause of **two minutes** during the build_phase of a simulation.   The pause happens immediately following the dumping of this line to stdout:
```
UVM_INFO @ 0.000 ns : uvma_cv32e40x_core_cntrl_agent.sv(81) uvm_test_top.env.core_cntrl_agent [CV32E40XCORECTRLAGT] Context handle is null; creating
```
This line is issued by [uvma_cv32e40x_core_cntrl_agent_c::get_and_set_cntxt()](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40x/env/uvme/uvma_cv32e40x_core_cntrl_agent.sv#L77).   There is no reason for this pause to occur, although I suspect that the parent class has a bug in its [uvma_core_cntrl_agent_c::build_phase(uvm_phase phase)](https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_core_cntrl/uvma_core_cntrl_agent.sv#L118).

Cadence Xcelium simulations of the cv32e40x environment do not exhibit this pause.  Note that DSIM and Xcelium simulation both produce the same simulation result, so this is not a _functional_ issue.

strichmo, 2021-08-02:
Could I get a branch and testcase where this occurs?  I could try to explore this using other simulators.

MikeOpenHWGroup, 2021-08-02:
(embarrassed cough)

Yes, "Steps to Reproduce" _is_ a key part of the issue template that _I_ created, isn't it?  Appreciate you keeping me honest.

This can be reproduce on the master branch of [core-v-verif](https://github.com/openhwgroup/core-v-verif/) when running any testcase:
```
$ git clone https://github.com/openhwgroup/core-v-verif.git
$ cd core-v-verif/cv32e40x/sim/uvmt
$ make sanity SIMULATOR=dsim
```
Key points:
1. This behavior manifests itself with the CV32E40**X**, but not the CV32E40**P**.
2. I only see the "pause" when running DSIM.
3. The simulations are functionally correct.  Pass/fail indication is not affected.
4. This is "important, but not urgent".  I am not gated in any significant manner by this issue.