openhwfoundation/core-v-verif #768: Update the CV32E40P environment to use the new OBI memory Agent

MikeOpenHWGroup, 2021-08-04:
### Task Outcome
As of #763 a new UVM agent, [obi_memory_agent](https://github.com/openhwgroup/core-v-verif/tree/master/lib/uvm_agents/uvma_obi_memory), intended to replace the module-based [mm_ram](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/tb/core/mm_ram.sv) model, has been integrated into the CV32E40X environment.  

The purpose of this issue is to determine whether this obi_memory_agent should be integrated into the CV32E40P environment.

PROS:
- All CV32E4 cores will use a common memory model.
- "Future proofs" development of new cores derived from the CV32E40P .

CONS:
- The latest version of the CV32E40P verification environment on core-v-verif will significantly differ from the one that was used to sign-off v1.0.0 of the CV32E40P core.
- Extra effort that does not benefit CV32E40P v1.0.0.

### Background information
The goal of the new agent is to support all current and future cores that use the [Open Bus Interface](https://github.com/openhwgroup/core-v-docs/blob/master/cores/obi/OBI-v1.2.pdf).

### Location Information
The `obi_memory_agent` is a library component of core-v-verif and can be found under core-v-verif/lib/uvm_agents.

### Completion Criteria
Full CV32E40P regression passes using the obi_memory_agent.

### Additional context
Implementing this will make the [mm_ram.sv](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/tb/core/mm_ram.sv) an orphan that is only used for Verilator simulations.  This _could_ be viewed as a benefit as it will allow for the removal of all those [`ifndef VERILATOR](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/tb/core/mm_ram.sv#L253) macro checks in the code.

strichmo, 2021-08-06:
I have created a PR that use the new UVM OBI agent: https://github.com/openhwgroup/core-v-verif/pull/773

Currently it is still undergoing full regression, but it does pass an Xcelium run of release check.

If further regressions pass I will enable merging of the PR, to be merged upon successful review.

strichmo, 2021-08-06:
The PR #773 is now merged.  The old UVM OBI agent is deprecated.  I am closing this issue.

closed 2021-08-06 by strichmo