openhwfoundation/core-v-verif #776: OVPSIM decrements minstret on debug trigger address match

strichmo, 2021-08-06:
### Type
* Functionally incorrect behavior suspected in the OVPSIM with the cv32e40p

### Steps to Reproduce
Please provide:
1. master branch of core-v-verif
2. This hash: 9d1130bae8cc2254f079e1d291a331dc914bb7f3
3. Use this command: 
```
% makeuvmt test TEST=debug_test_trigger CV_CORE=cv32e40p SEED=1
```
The test should repeat with other simulators.  It was observed in both Xcelium and DSIM.

The test failure:
```
UVM_ERROR @ 10398.300 ns : uvmt_cv32e40p_step_compare.sv(125) reporter [Step-and-Compare] GPR[7] expected=0x00000226 and actual=0x00000227 PC=0x1a11082c
UVM_ERROR @ 10419.300 ns : uvmt_cv32e40p_step_compare.sv(125) reporter [Step-and-Compare] GPR[7] expected=0x00000226 and actual=0x00000227 PC=0x1a110830
UVM_ERROR @ 10434.300 ns : uvmt_cv32e40p_step_compare.sv(125) reporter [Step-and-Compare] GPR[7] expected=0x00000226 and actual=0x00000227 PC=0x1a110834
```

The miscompare is a _minstret_ CSR read that is off by one.  Upon waveform inspection, it appears that OVPSIM is actually decrementing minstret internally when the instruction that is triggering debug is "executed".  _timing_ for triggers are forced to 0 in all CORE-V E* cores, so the trigger should take effect without executing that instruction.

[image]

PilotAware, 2021-08-09:
Steve, did you say this passes on cv32e40x ?
I tried 
makeuvmt test TEST=debug_test_trigger CV_CORE=cv32e40x SEED=1

and I got a different failure

eroom1966, 2021-08-09:
Steve, did you say this passes on cv32e40x ?
I tried
makeuvmt test TEST=debug_test_trigger CV_CORE=cv32e40x SEED=1

and I got a different failure

strichmo, 2021-08-10:
> Steve, did you say this passes on cv32e40x ?
> I tried
> makeuvmt test TEST=debug_test_trigger CV_CORE=cv32e40x SEED=1
> 
> and I got a different failure, I do see the backward counting which looks odd - I will investigate further

@eroom1966 The test does not fully pass on the current cv32e40x, but it should proceed far enough to see an debug trigger match which enters debug mode.  This debug mode entry does not increment minstret RVVI CSR as in the cv32e40p.

eroom1966, 2021-08-10:
@strichmo 
I still have the issue we discussed previously regarding the ram location, so this is not possible to debug with the latest code.

-This contains the memory here 

interface RVVI_memory;
    reg [31:0] mem [bit[29:0]];
endinterface

I am not sure why the update/merge has not worked correctly, I will try to revert some changes and highlight the issue, but we need to fix this issue where we have the memory interface in the CPU SV wrapper

strichmo, 2021-08-12:
This is now fixed on the master branch and will shortly be propagated to cv32e40p/dev and release branches.

closed 2021-08-12 by strichmo