openhwfoundation/core-v-verif #791: vplan review for RV32Zicsr extension

strichmo, 2021-08-10:
### Task Outcome
Updated vplan coverage plan definition for the RV32 M extension.  This vplan should be applicable to all CORE-V processors implementing the Zicsr extension (i.e. all cores)

### Background information
This review will update the coverage definition using the new ISACOV UVM coverage agent in all CORE-V testbenches.

### Location Information
The updated excel can be fetched from this branch:
https://github.com/strichmo/core-v-verif/blob/strichmo/pr/vplan_isacov_abz_etc/docs/VerifPlans/ISA/RV32/Simulation/RV32Zicsr_Extension_Instruction.xlsx

### Completion Criteria
Vplan meeting is completed.
All action items completed
Vplan definition updated on master branch.

silabs-robin, 2021-08-12:
Should we let the Zicsr vplan include something about instret and other counters? There is one paragraph in the Zicsr spec that discusses this.

silabs-robin, 2021-08-12:
For reference, this is the part I was thinking about (picture below).
It might conceivably have an impact on i.a. any of the performance counters.
I don't mean to prescribe any recommended course of action, but I want to check if we are safe from any impact relating to this.

[image]

Silabs-ArjanB, 2021-08-12:
I don't think I fully understand your question, but the above should be verified as well for minstret (whether it is part of Zicsr or not I will leave to Steve). We do have a directed performance counter test that does already check this I believe though.

strichmo, 2021-08-28:
Zicsr updated with: https://github.com/openhwgroup/core-v-verif/pull/830

closed 2021-08-28 by strichmo