openhwfoundation/core-v-verif #792: vplan review for RV32Zifencei extension

strichmo, 2021-08-10:
### Task Outcome
Updated vplan coverage plan definition for the RV32 M extension.  This vplan should be applicable to all CORE-V processors implementing the I extension (i.e. all cores)

### Background information
This review will update the coverage definition using the new ISACOV UVM coverage agent in all CORE-V testbenches.

### Location Information
The updated excel can be fetched from this branch:
https://github.com/strichmo/core-v-verif/blob/strichmo/pr/vplan_isacov_abz_etc/docs/VerifPlans/ISA/RV32/Simulation/RV32Zifencei_Extension_Instruction.xlsx

### Completion Criteria
Vplan meeting is completed.
All action items completed
Vplan definition updated on master branch.

silabs-robin, 2021-08-12:
Hehe that was a short vplan.

I have one question.
Shouldn't the vplan desire more than merely executing fencei?
It seems the 40x implements fencei by a jump. So I get that the pipeline will be flushed, but will necessarily the prefetcher (and everything related) also get flushed?
I don't fully understand all of this, but maybe there should be a directed test a la: planning to do an ADDI, but first we store to the addi's location (maybe another addi with different immediate), and we see how fencei / no fencei impacts this as expected?
Is this too uarch-specific?

Silabs-ArjanB, 2021-08-12:
We should specifically verify the following:

- Ability to block fence(i) completion by holding off fence handshake
- Tolerance against random fence ack signal when no fence(i) instruction is being executed (easy to check with Formal)
- Full flush (including suppression of data_req_o) of all pipeline stages and prefetcher on fence(i) (which is more than required by RISC-V)
- Interaction of fence handshake interface with load/store interface (and to be added write buffer and to be added extension interface)

strichmo, 2021-08-28:
Comments above will be rolled into a fence.i micro-architectural vplan.

This issue will be closed with the ISA vplan: https://github.com/openhwgroup/core-v-verif/pull/830

closed 2021-08-28 by strichmo