openhwfoundation/core-v-verif #793: vplan review for RV32 sequential execution of instructions

strichmo, 2021-08-10:
# Template for CORE-V-VERIF Task Issues
### Task Outcome
Updated vplan coverage plan definition ISA RV32 instruction sequences.

### Background information
This review will update the coverage definition using the new ISACOV UVM coverage agent in all CORE-V testbenches.

### Location Information
The updated excel can be fetched from this branch:
https://github.com/strichmo/core-v-verif/blob/strichmo/pr/vplan_isacov_abz_etc/docs/VerifPlans/ISA/RV32/Simulation/RV32ISA_Sequential.xlsx

### Completion Criteria
Vplan meeting is completed.
All action items completed
Vplan definition updated on master branch

MikeOpenHWGroup, 2021-08-10:
Hi @strichmo, I took the liberty of stripping the templated title block from this issue.

strichmo, 2021-08-28:
Reviewed and updated in this PR: https://github.com/openhwgroup/core-v-verif/pull/830

closed 2021-08-28 by strichmo