openhwfoundation/core-v-verif #794: vplan review for RV32 counters for cv32e40x

strichmo, 2021-08-10:
### Task Outcome
Updated vplan coverage plan definition for the RV32 M extension.  This vplan should be applicable to all CORE-V processors implementing the I extension (i.e. all cores)

### Background information
This review will update the coverage definition using the new ISACOV UVM coverage agent in all CORE-V testbenches.

### Location Information
The updated excel can be fetched from this branch:
https://github.com/strichmo/core-v-verif/blob/strichmo/pr/vplan_isacov_abz_etc/cv32e40x/docs/VerifPlans/Simulation/privileged_spec/CV32E40X_Counters.xlsx

### Completion Criteria
Vplan meeting is completed.
All action items completed
Vplan definition updated on master branch.

silabs-robin, 2021-08-12:
I don't have any feedback per se, but I have some questions.

Should countinhibit (and maybe others too) have some assertions?
Why is minstret link to coverage waived?
How are we sure that the items with coverage method "testcase" should not have "functional coverage"?
Are all "link to coverage" cells intended to be filled out as implementation progresses?

strichmo, 2021-09-29:
Resolving this particular review.  There will be another to update for E40X-specific counter events.

closed 2021-09-29 by strichmo