openhwfoundation/core-v-verif #807: on RVFI and RVVI a triggered instruction should not be considered valid

strichmo, 2021-08-12:
### Type
* Functionally incorrect behavior

The RVVI and RVFI are both considering a "triggered" transaction to be a valid transaction.  However the E40X (and E40S and E40P) all fix the "timing" setting to 0, which indicates that the instruction is not to be executed before entering debug mode.  Thus the instruction should not be "valid" and should not increment order on either interface.

### Steps to Reproduce
From the master branch: 
% git co master
# Commit: 814d17050449d068fe05fc92b2138ddc00c3c64d
% makeuvmt test TEST=debug_test_trigger CV_CORE=cv32e40x 

### Additional context
This will need to be fixed both in the OVPSIM model and the current RVFI implementation.

eroom1966, 2021-08-13:
@strichmo 
this should be fixed in this PR https://github.com/openhwgroup/core-v-verif/pull/809
bear in mind the test above will fail until the RVFI is corrected in the same manner

strichmo, 2021-08-28:
Closing this issue and opening another one to track that we have master broken for the cv32e40x and that we need a reversion on master.

closed 2021-08-28 by strichmo