openhwfoundation/core-v-verif #810: Intermittent failure of debug_test on CV32E40X

MikeOpenHWGroup, 2021-08-14:
I have observed that the `debug_test` fails intermittently.  I suspect it is an environment, not RTL, issue.

### Type
Functionally incorrect behaviour

### Steps to Reproduce
1. checkout core-v-verif branch `cv32e40x/dev`, hash: `6fe8e00d`
2. cd core-v-verif/cv32e40x/sim/uvmt
3. make test TEST=debug_test SIMULATOR=dsim DSIM_SEED=333967796

### Additional context
If you are not using DSIM version 20210621.3.1, the seed provided in the above example is probably useless.

Failure signature:
```
Irq check done
-------------------------
Test 21: Setting stopcount bit=1
Wait for Debugger
------------------------
Test 18: Single stepping
UVM_ERROR @ 1722987.300 ns : uvme_cv32e40x_core_sb.sv(366) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 67948, csr: dpc, rvfi = 0x000036c2, rvvi = 0x000036be
UVM_ERROR @ 1730190.300 ns : uvme_cv32e40x_core_sb.sv(366) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 68282, csr: dpc, rvfi = 0x000036cc, rvvi = 0x000036c8
1732431.000 ns: Illegal instruction (core 0) at PC 0x000036cc:
UVM_ERROR @ 1739325.300 ns : uvme_cv32e40x_core_sb.sv(366) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 68700, csr: dpc, rvfi = 0x00000cdc, rvvi = 0x00000cda
UVM_ERROR @ 1741491.300 ns : uvme_cv32e40x_core_sb.sv(366) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 68801, csr: dpc, rvfi = 0x00000cde, rvvi = 0x00000cdc
UVM_ERROR @ 1743819.300 ns : uvme_cv32e40x_core_sb.sv(366) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 68902, csr: dpc, rvfi = 0x00000ce0, rvvi = 0x00000cde
UVM_INFO @ 1743819.300 ns : uvme_cv32e40x_core_sb.sv(190) uvm_test_top.env.core_sb [CORESB] checked 68902 instruction retirements
UVM_INFO @ 1743819.300 ns : uvme_cv32e40x_core_sb.sv(191) uvm_test_top.env.core_sb [CORESB] checked 36489 GPR updates
UVM_INFO @ 1743819.300 ns : uvme_cv32e40x_core_sb.sv(192) uvm_test_top.env.core_sb [CORESB] checked 1584725 CSRs
UVM_INFO @ 1743819.300 ns : uvm_report_server.svh(847) reporter [UVM/REPORT/SERVER]
```

strichmo, 2021-08-17:
These failures have been triaged to an RVFI implementation issue: https://github.com/openhwgroup/cv32e40x/issues/187

strichmo, 2021-08-25:
Based on recent RTL and testbench work, the debug_test now appears to be very stable, even with random stalls added.  Therefore the intent of this issue was resolved.

This bug was resolved in relation to this issue:
https://github.com/openhwgroup/cv32e40x/issues/187

closed 2021-08-25 by strichmo

MikeOpenHWGroup, 2021-08-25:
Fix confirmed.