openhwfoundation/core-v-verif #841: OVPSIM to does not recognize write to mcause[5]

strichmo, 2021-09-01:
### Type
* Functionally incorrect behavior

In the CV32E40X, the mcause exception field was extended to 6 bits to support the following new exception code:
- instruction bus fault (48)

In this test mcause is being written with 0xa5a5a5a5.  In the RVVI[mcause] CSR, the value is stored as 0x8000_0005 instead of 0x8000_0025 as in the RTL.

```
UVM_ERROR @ 90.300 ns : uvme_cv32e40x_core_sb.sv(370) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 4, csr: mcause, rvfi = 0x80000025, rvvi = 0x80000005
```

### Steps to Reproduce
Please provide:
1. Use main core-v-verif repo: https://github.com/openhwgroup/core-v-verif.git
2. Use this commit on cv32e40x/dev branch: 16969eb3966094aea83bdba2d70fc91616cc69ab
3. makeuvmt test TEST=cv32e40x_csr_access_test CV_CORE=cv32e40x

silabs-hfegran, 2021-09-10:
This field will be expanded to 8 bits for 40x

Silabs-ArjanB, 2021-09-10:
Same for 40S

strichmo, 2021-09-10:
Thanks for comments @Silabs-ArjanB @silabs-hfegran .

I am updating the hash and testcase for this to include mcause [7:5] as missing in the OVPSim model.

eroom1966, 2021-09-13:
@strichmo 
Hmm this is an interesting one
in the specification is says that bits mcause[7:0] are RW, this was an oversight in the model which you quite rightly point out was still the defalt 5 bits.
I changed the ecode mask by adding 
--override root/cpu/ecode_mask=0xff

but the compare still fails - this time the RTL does not allow the writing of mcause[7]

Info 4: 'root/cpu', 0x000000000000008a(_start_main+8): 342210f3 csrrw   ra,mcause,tp
Info   mcause 00000000 -> 800000a5
UVM_ERROR @ 108.300 ns : uvme_cv32e40x_core_sb.sv(370) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 4, csr: mcause, rvfi = 0x80000025, rvvi = 0x800000a5

CSR Write mcause = 0xa5a5a5a5
RTL = 0x80000025
RM = 0x800000a5

I will change the default in the base model to be an ecode mask of 0xff
but this still seems wrong ?

silabs-hfegran, 2021-09-13:
@eroom1966 The RTL update to address this was merged in https://github.com/openhwgroup/cv32e40x/pull/221 last friday - are you on the latest dev?

eroom1966, 2021-09-13:
@strichmo All my tests have passed, I do see we are now getting a WARNING of the form

UVM_WARNING @ 0.000 ns : uvm_traversal.svh(267) reporter [UVM/COMP/NAME] the name "monitor`" of the component "uvm_test_top.env.fencei_agent.monitor`" violates the uvm component name constraints

should I issue a PR ?

strichmo, 2021-09-13:
@eroom1966 I'm not sure exactly what it is complaining about but I just wrote the UVM agent so I will take that one on.  Please go ahead and file PR if it passes.

eroom1966, 2021-09-13:
I think I may have been missing some updates, I performed another pull/merge and this message has disappeared
All tests on E40X pass

strichmo, 2021-09-13:
Sounds good...I'm not seeing the warning as well.  I did fix an issue similar to this about a week ago.

eroom1966, 2021-09-13:
I think redoing the merge also resolved the RTL issue of mcause[7] not writable

strichmo, 2021-09-13:
Fix is merged with this PR: https://github.com/openhwgroup/core-v-verif/pull/857

This PR fully fixes the expected CSR values in the generated CSR access test: https://github.com/openhwgroup/core-v-verif/pull/859

closed 2021-09-13 by strichmo