openhwfoundation/core-v-verif #860: OVPSIM repeats error when exception vector has bus fault

strichmo, 2021-09-13:
### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. Use cv32e40x/dev branch of core-v-verif: https://github.com/openhwgroup/core-v-verif.git
2. Use this commit: 8b4b03039a622b06189dbdc3fd92602fcc0a3818
3. makeuvmt test TEST=corev_rand_instr_obi_err WAVES=1 ADV_DEBUG=1 CV_CORE=cv32e40x SEED=550659836  USER_RUN_FLAGS=+UVM_MAX_QUIT_COUNT=50,NO

Note that this testcase was created with Xcelium version 21.07.a001.  It is likely another version will need another seed, but that should be easy to find.  Please ask @strichmo if a seed for a particular seed for a particular Xcelium is required.

The failure occurs when an instruction bus error is detected on the first instruction of the exception table (mtvec).  When this occurs, it appears the OVPSIM will continuously re-enter the exception.  **It does not refetch the instruction and simply repeats the mtvec instruciton.**  It would desirable to have OVPSim refetch the instruction so that the testbench could signal a "valid" instruction fetch and the test could proceed.

[image]

eroom1966, 2021-09-16:
@strichmo 
Sorry for the delay in looking at this.
I am unable to reproduce 
I run the following

$ makeuvmt CV_CORE=cv32e40x clean_all
$ makeuvmt test TEST=corev_rand_instr_obi_err WAVES=1 ADV_DEBUG=1 CV_CORE=cv32e40x SEED=550659836 USER_RUN_FLAGS=+UVM_MAX_QUIT_COUNT=50,NO

I am getting this error message
...
/home/moore/riscv/corev-openhw-gcc-ubuntu1804-20200913/bin/riscv32-corev-elf-gcc  \
	-Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic \
	-I ../../tests/asm \
	-o /home/moore/git/openhw_e40x/core-v-verif/cv32e40x/tests/programs/corev-dv/corev_rand_instr_obi_err/corev_rand_instr_obi_err_0.elf \
	-nostartfiles \
	 \
	-T /home/moore/git/openhw_e40x/core-v-verif/cv32e40x/bsp/link.ld \
	-L /home/moore/git/openhw_e40x/core-v-verif/cv32e40x/bsp \
	-lcv-verif
/home/moore/riscv/corev-openhw-gcc-ubuntu1804-20200913/bin/../lib/gcc/riscv32-corev-elf/11.0.0/../../../../riscv32-corev-elf/bin/ld: /home/moore/git/openhw_e40x/core-v-verif/cv32e40x/bsp/libcv-verif.a(crt0.o): in function `.L0 ':
/home/moore/git/openhw_e40x/core-v-verif/cv32e40x/bsp/crt0.S:46: undefined reference to `main'
collect2: error: ld returned 1 exit status

we can share a desktop if you have time ?

eroom1966, 2021-09-16:
I recall this is similar to a previous Issue, I see no test files in the directory
cv32e40x/tests/programs/corev-dv/corev_rand_instr_obi_err

strichmo, 2021-09-16:
My apologies.  I forgot the targets for corev-dv in the above command.

```
% makeuvmt corev-dv gen_corev-dv test TEST=corev_rand_instr_obi_err WAVES=1 ADV_DEBUG=1 CV_CORE=cv32e40x SEED=550659836 USER_RUN_FLAGS=+UVM_MAX_QUIT_COUNT=50,NO
```

strichmo, 2021-09-16:
Please let me know which Xcelium version you are using if this does not repeat.  I should be able to dig up the version here and generate another seed if necessary.

MikeOpenHWGroup, 2021-09-16:
Test in `$(CV_CORE_LC)/tests/programs/corev-dv` are generated by corev-dv, our extension of the Google riscv-dv random instruction stream generator.  So unless you run the generator there will not be a test-program at that location.

See [corev-dv generated tests](https://github.com/openhwgroup/core-v-verif/tree/master/mk/uvmt#corev-dv-generated-tests) in the common Makefiles director for details (its trivial).

eroom1966, 2021-09-16:
> 
> 
> Please let me know which Xcelium version you are using if this does not repeat. I should be able to dig up the version here and generate another seed if necessary.

I am running CDNS-UVM-1.2-ML (20.03-s009)

I think I am reproducing the issue, and I do have a clue what it could be

eroom1966, 2021-09-16:
Because the model fetches occur in 2-byte accesses (in order to support Compressed instructions), but the bus Instruction width is 4-bytes, I put a cache match into the bus wrapper, so it did not have to fetch the upper half word again, after requesting the lower half word.

when the busfault was added in, I should have changed this, to invalidate the fetched (cached) value, if a fault occurs
very simple fix

Testcase now passes - I will submit a PR

eroom1966, 2021-09-16:
Please refer to 
https://github.com/openhwgroup/core-v-verif/pull/875

strichmo, 2021-09-28:
Issue is resolved.

closed 2021-09-28 by strichmo