openhwfoundation/core-v-verif #889: vplan review for fence.i for cv32e40x

silabs-robin, 2021-09-22:
### Task Outcome
The verification plan for `fence.i` functionality is reviewed for CV32E40X.

### Background information
.

### Location Information
The new spreadsheet can be fetched from this branch:
https://github.com/silabs-robin/core-v-verif/blob/fencei_vplan/cv32e40x/docs/VerifPlans/Simulation/micro_architecture/CV32E40X_fencei.xlsx

### Completion Criteria
The accepted review criteria are completed and the final reviewed vplan is placed in the above directory.

Silabs-ArjanB, 2021-09-22:
Hi @silabs-robin ,

Here is my feedback on the (great) vplan:

- The X interface currently can only signal that there is an offloaded load/store. It just signals 'load _or_ store', so the implementation of a CPU will also have to make sure that offloaded loads complete before a fence.i (which is not a specification requirement). If you think this is an issue, please let me know and then we add separate load and store signals in the X interface. 
- The 'ack withold' feature can easily be checked with an assertion.
- It would be good to check that a fence.i that is in the shadow of a taken branch or jump (or interrupt or debug entry) is fully killed and so does not raise the request signal (this could also be in the interrupt vplan as similar checks are needed for all possible side effects of instructions that should not occur if an instruction is supposed to be killed).
- With respect to the 'multi-cycle' feature: I am not aware why the cycle count should not be deterministic.
- With respect to the 'flush' feature. I don't like the 'data_req_o is eventually suppressed'. If there is a load, store, offloaded load, offloaded store present in the pipeline when the fence.i reaches WB, then data_req_o should be immediately suppressed (it is not allowed to be high for any cycle).

silabs-robin, 2021-09-23:
@Silabs-ArjanB 

> The X interface currently can only signal that there is an offloaded load/store

Understood. I can't really foresee enough of the consequences of either way.
So I think it is fine to leave it as is and see how it goes.

> The 'ack withold' feature can easily be checked with an assertion

True. Changing it.

> It would be good to check that a fence.i that is in the shadow of a taken branch or jump [...]

I'll add it.

> With respect to the 'multi-cycle' feature: I am not aware why the cycle count should not be deterministic.

I can specify [when it is 2 or 3 cycles](https://cv32e40x-user-manual.readthedocs.io/en/latest/pipeline.html).

> With respect to the 'flush' feature. I don't like the 'data_req_o is eventually suppressed'

Is it not possible that one store is ongoing and there is still one in the write buffer that must be allowed to complete?

Silabs-ArjanB, 2021-09-23:
> 
>     With respect to the 'flush' feature. I don't like the 'data_req_o is eventually suppressed'
> 
> Is it not possible that one store is ongoing and there is still one in the write buffer that must be allowed to complete?

You are right.

strichmo, 2021-09-23:
Thanks @silabs-robin , great detail here.

- Regarding "ack withold", yes there should be assertion.  However it would be good to determine and document functional coverage for the maximum latency we will use in simulation.  There should be a coverpoint for [0:max] cycles.
- I would like to have some check from the OBI monitors to ensure that OBIs are idle before fencei_flush_req_o is asserted.  (Of course this is in addition to a write buffer check as well).
- The self-modifying instruction check is a good one.  For maximum effectiveness it would be nice if the write occurred external to the core.  We should be able to rather easily create a virtual peripheral that can load an instruction to both memories (ISS and OBI agent) upon a flush_req to ensure that the old prefetch bufferer isn't used for instruction pc[fence.i] + 4.

silabs-robin, 2021-09-24:
@strichmo 

> [ack withold] it would be good to determine and document functional coverage for the maximum latency

I added a sentence asking for coverage to check at least {[0:5]}. (Confered with Arjan)

> I would like to have some check from the OBI monitors to ensure that OBIs are idle before fencei_flush_req_o is asserted

Would this be an assertion or something else? And are we talking about uvma_obi_memory_mon? If it is an assertion then the agent's vif must have some output signals that are driven early enough that it can compare against the flush interface?

> [self-modifying instruction check] it would be nice if the write occurred external to the core

Should this be in addition to a dv stream that checks it "manually" with stores, or should this replace and be in stead of such a dv stream?

silabs-robin, 2021-09-30:
Review done, plan merged