openhwfoundation/core-v-verif #896: CV32E40P Test Program is not Compiling

drac98, 2021-09-26:
I'm new to RISC-V and I have been trying to run a simulation of CV32E40P with Verilator.

## CV32E40P Test Programs Compiling Error

I followed the instructions given [here](https://github.com/openhwgroup/core-v-verif/tree/2f8d8d3a4413b87e95fc4f399ff711475473e41d/cv32e40p/sim/core).
- Point your environment variable RISCV to your RISC-V toolchain. Call make to run the default test (hello_world).

Compiler : `corev-openhw-gcc-ubuntu2004-20200913`
Verilator : `4.212 2021-09-01 rev v4.212-20-g27c08a65`

### Compile error

Terminal output:
https://pastebin.com/b67RYu6K

- [x] Cloning CV32E40P RTL model
- [x] Compiling CORE TB and CV32E40P with Verilator
- [x] Compiling BSP
- [ ] Compiling test-program ../../tests/programs/custom/hello-world/hello-

### Steps to Reproduce

1. Install recommended toolchain
2. Clone the `core-v-verifa` repo
2. go to `CV32E40P/sim/core` directory
3. run `make`

If more information is needed, please let me know.

MikeOpenHWGroup, 2021-09-27:
Hi @drac98, thanks for bringing this issue to our attention.  I have been able to reproduce the issue and am working on a fix.

MikeOpenHWGroup, 2021-09-28:
Hi @drac98, with #899 merged in I believe things should be working for you now.  Please check and let me know.   FYI, this fix is on the master branch and will be propagated to the `cv32e40p/dev` and `cv32e40p/release` branches at the end of this week.

drac98, 2021-09-28:
Hello @MikeOpenHWGroup , the fix works. Thank you.

closed 2021-09-28 by drac98