openhwfoundation/core-v-verif #904: OVPSIM violates debug 1.0.0 spec when interrupt occurs during step

strichmo, 2021-09-28:
### Type
* Functionally incorrect behavior

### Steps to Reproduce
1. https://github.com/strichmo/core-v-verif/tree/strichmo/temp/ovpsim_debug_step_intr
2. c969738d21f832a9cb061b24968e7ef2f599c264
(Note that this hash points to a branch in @silabs-oysteink 's fork.
3. makeuvmt test TEST=debug_test

The failure occurs when interrupts occur during debug mode single-stepping:

```
UVM_ERROR @ 3339468.300 ns : uvme_cv32e40x_core_sb.sv(305) uvm_test_top.env.core_sb [CORESB] PC Mismatch, rvfi_order: 300150, rvvi_order: 300150, rvfi.pc = 0x1a110800, rvvi.pc = 0x00000078
UVM_ERROR @ 3339468.300 ns : uvme_cv32e40x_core_sb.sv(311) uvm_test_top.env.core_sb [CORESB] INSN Mismatch, order: 300150, rvfi.insn = 0x7b251073, rvvi.insn = 0x2490006f
UVM_ERROR @ 3339468.300 ns : uvme_cv32e40x_core_sb.sv(370) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 300150, csr: dpc, rvfi = 0x00000078, rvvi = 0x0000cc1e
UVM_ERROR @ 3339480.300 ns : uvme_cv32e40x_core_sb.sv(305) uvm_test_top.env.core_sb [CORESB] PC Mismatch, rvfi_order: 300151, rvvi_order: 300151, rvfi.pc = 0x1a110804, rvvi.pc = 0x1a110800
UVM_ERROR @ 3339480.300 ns : uvme_cv32e40x_core_sb.sv(311) uvm_test_top.env.core_sb [CORESB] INSN Mismatch, order: 300151, rvfi.insn = 0x00001517, rvvi.insn = 0x7b251073
UVM_INFO @ 3339480.300 ns : uvme_cv32e40x_core_sb.sv(190) uvm_test_top.env.core_sb [CORESB] checked 300151 instruction retirements
```
The OVPSIM is being overridden into debug 1.0.0 mode.

```
    --override root/cpu/debug_version=1.0.0-STABLE (Enumeration) (default=0.13.2-DRAFT) (override) Specify required Debug Architecture version
```

According to the 1.0.0 debug specification (p. 50)

```
If control is transferred to a trap handler while executing the instruction, then Debug Mode is
re-entered immediately after the PC is changed to the trap handler, and the appropriate tval and
cause registers are updated. In this case none of the trap handler is executed, and if the cause was
a pending interrupt no instructions might be executed at all.
```

The RTL in Oystein's branch has been updated to adhere to the above paragraph.  It appears that OVPSIM is actually retiring the first instruction of the trap handler instead of immediately re-entering debug as described above.

[image]

### Additional context
Add any other context about the problem here.

eroom1966, 2021-10-08:
@silabs-oysteink @strichmo 

I do not see the exact same error reported here.
To be precise these are the exact commands I have executed 

git clone https://github.com/strichmo/core-v-verif -b strichmo/temp/ovpsim_debug_step_intr
makeuvmt CV_CORE=cv32e40x clean_all
makeuvmt test CV_CORE=cv32e40x TEST=debug_test

The failure I see is as follows

> UVM_INFO @ 3743361.300 ns : uvme_cv32e40x_core_sb.sv(316) uvm_test_top.env.core_sb [CORE_SB] Compared 330000 instructions
> UVM_INFO @ 3853176.300 ns : uvme_cv32e40x_core_sb.sv(316) uvm_test_top.env.core_sb [CORE_SB] Compared 340000 instructions
> 
> UVM_INFO @ 3963435.300 ns : uvme_cv32e40x_core_sb.sv(316) uvm_test_top.env.core_sb [CORE_SB] Compared 350000 instructions
> UVM_ERROR @ 3982461.300 ns : uvme_cv32e40x_core_sb.sv(305) uvm_test_top.env.core_sb [CORESB] PC Mismatch, rvfi_order: 351754, rvvi_order: 351754, rvfi.pc = 0x1a110800, rvvi.pc = 0x00000078
> UVM_ERROR @ 3982461.300 ns : uvme_cv32e40x_core_sb.sv(311) uvm_test_top.env.core_sb [CORESB] INSN Mismatch, order: 351754, rvfi.insn = 0x7b251073, rvvi.insn = 0x2490006f
> UVM_ERROR @ 3982461.300 ns : uvme_cv32e40x_core_sb.sv(370) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 351754, csr: dpc, rvfi = 0x00000078, rvvi = 0x00003848
> UVM_ERROR @ 3982473.300 ns : uvme_cv32e40x_core_sb.sv(305) uvm_test_top.env.core_sb [CORESB] PC Mismatch, rvfi_order: 351755, rvvi_order: 351755, rvfi.pc = 0x1a110804, rvvi.pc = 0x1a110800
> UVM_ERROR @ 3982473.300 ns : uvme_cv32e40x_core_sb.sv(311) uvm_test_top.env.core_sb [CORESB] INSN Mismatch, order: 351755, rvfi.insn = 0x00001517, rvvi.insn = 0x7b251073
> UVM_INFO @ 3982473.300 ns : uvme_cv32e40x_core_sb.sv(190) uvm_test_top.env.core_sb [CORESB] checked 351755 instruction retirements
> UVM_INFO @ 3982473.300 ns : uvme_cv32e40x_core_sb.sv(191) uvm_test_top.env.core_sb [CORESB] checked 219403 GPR updates
> UVM_INFO @ 3982473.300 ns : uvme_cv32e40x_core_sb.sv(192) uvm_test_top.env.core_sb [CORESB] checked 8090342 CSRs
> UVM_INFO @ 3982473.300 ns : uvm_report_catcher.svh(705) reporter [UVM/REPORT/CATCHER] 

please advise if I am missing a step, it looks like the timing is different - maybe that does not matter ?

Thx
Lee

silabs-oysteink, 2021-10-08:
It looks to me like you are reproducing the same issue, although with slightly different timing. The case is when an interrupt is taken during single step, which results in no retired instructions before re-entering debug mode. RVFI PC points at the first instruction of debug (0x1a110800), while RVVI points at the first instruction of the interrupt handler (0x78). My understanding of debug spec V1.0.0 is that the RTL is correct here, no instruction retired during step, mepc updated according to the taken interrupt and the PC of the first instruction of the interrupt handler is stored to dpc.

eroom1966, 2021-10-08:
reproduced issue and are investigating WRT 1.0.0 specification

eroom1966, 2021-10-08:
I have a fix for this, my intent was to commit this back to the cv32e40x/dev branch - is that OK ?

eroom1966, 2021-10-08:
Hi Steve
I am holding off this commit, it breaks the existing debug_test on this branch for 0.13.2 / 0.14 
is that expected ?

silabs-oysteink, 2021-10-11:
@eroom1966  Could you elaborate on what is failing for 0.13.2/0.14, and how? Is the RTL failing on 'stock' ISS 0.13.2, or after the fix for step+interrupt is done for 1.0?

eroom1966, 2021-10-11:
still investigating, but this behavior differs on the RTL used in the CV32E40X/dev 
We think the behavior of 1.0.0 is correct, and 0.13.2/0.14 was incorrect

We changed something in the reference based upon an interpretation in the spec to match your RTL, and now we are reverting it to the original behavior

I need to trace back through our email exchanges to determine why we were asked to change this

in essence we have flip-flopped the behavior

eroom1966, 2021-10-11:
@silabs-oysteink 
you will notice that the debug test passes for e40x but fails for e40p, I think the e40p still has the RTL code bug

strichmo, 2021-10-19:
The issue is fixed in the cv32e40x and the ISS model.  The above issue (669) will clarify behavior in the CV32E40P.  Therefore this issue will be resolved.