openhwfoundation/core-v-verif #997: Wrong number of arguments specified for 'interrupt' attribute

MikeOpenHWGroup, 2021-10-31:
### Type
* Compile error

### Steps to Reproduce
1. Check out hash `7f9ff9881fd399af9f786a0aa5bf1ee0b5ab54a4` of core-v-verif (current head of `cv32e40x/dev`).
2. $ cd core-v-verif/cv32e40x/sim/uvmt
3. $ make test TEST=interrupt_test

### Error message from [Metrics log](https://openhwgroup.metrics.ca/logs?externalIds[]=2519536f-f7a9-4295-853e-ff5d721c0f4d):
```
/opt/riscv/bin/riscv32-unknown-elf-gcc \
	 \
	-Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic \
	 \
	-I ../../tests/asm \
	-I /mux-flow/build/repo/cv32e40x/bsp \
	-o /mux-flow/build/results/default/interrupt_test/0/test_program/interrupt_test.elf \
	-nostartfiles \
	/mux-flow/build/repo/cv32e40x/tests/programs/custom/interrupt_test/interrupt_test.c \
	-T /mux-flow/build/repo/cv32e40x/bsp/link.ld \
	 -L /mux-flow/build/results/default/interrupt_test/0/test_program/bsp \
	-lcv-verif

In file included from /mux-flow/build/repo/cv32e40x/tests/programs/custom/interrupt_test/interrupt_test.c:5:0:
/mux-flow/build/repo/cv32e40x/tests/programs/custom/interrupt_test/interrupt_test.h:73:1: error: wrong number of arguments specified for 'interrupt' attribute
 __attribute__((interrupt ("machine"))) void m_software_irq_handler(void);
 ^~~~~~~~~~~~~

/mux-flow/build/repo/cv32e40x/tests/programs/custom/interrupt_test/interrupt_test.h:74:1: error: wrong number of arguments specified for 'interrupt' attribute
 __attribute__((interrupt ("machine"))) void m_timer_irq_handler(void);
 ^~~~~~~~~~~~~
```

strichmo, 2021-11-01:
@MikeOpenHWGroup I seem to remember this being an issue with using an older toolchain.  Could you show me which version of gcc you are using?

MikeOpenHWGroup, 2021-11-01:
gcc 9.3.0 on my Ubuntu20.04 desktop and 4.8.5 on CMC's CentOS7 machines.

strichmo, 2021-11-01:
Do you see the error with 9.3.0?  I do not see it with 9.3.0 on my systems.

MikeOpenHWGroup, 2021-11-01:
Ah, shoot!  9.3.0 is my desktop's gcc.  My toolchain's gcc (i.e. `/opt/riscv/bin/riscv32-unknown-elf-gcc`) is 7.1.1.  Guess I'll need to rebuild my toolchain.

strichmo, 2021-11-02:
@MikeOpenHWGroup I'm going to set Resolved and assign to you to close.  Please re-assign and let me know if any more action is required.

MikeOpenHWGroup, 2021-11-04:
Agreed, this can be closed.  I updated my riscv toolchain to the top-of-tree on the [Embecosm toolchain downloads website](https://www.embecosm.com/resources/tool-chain-downloads/) (great resource, that) and the test compiles and runs as expected.  For the record the riscv toolchain I am now using is `riscv32-embecosm-gcc-ubuntu2004-20211027` which is based on `gcc version 12.0.0 20211027`.

Note that the riscv toolchain used by MetricsCI also needs to be updated, see Metrics support ticket #1136.

closed 2021-11-04 by MikeOpenHWGroup