openhwfoundation/core-v-verif #1030: Should we suppress ReadingOutputModport dsim warning?

MikeOpenHWGroup, 2021-11-20:
[CV32E40X issue #259](https://github.com/openhwgroup/cv32e40x/issues/259) describes a ReadingOutputModport compile-time warning emitted by Metrics dsim.   This warning is not seen with Cadence Xcelium and in a discussion with Metrics it was agreed that reading from an output modport does not violate the LRM.

I would like to propose that we suppress this dsim warning, but will only do so if @strichmo agrees it is safe to so.

strichmo, 2021-11-21:
Given that Xcelium does not issue a warning, I am OK with suppressing this warning.

I will reassign back to @MikeOpenHWGroup to make the change.

MikeOpenHWGroup, 2021-11-23:
This issue can be closed when pull-request #1034 is merged in.

closed 2021-11-24 by MikeOpenHWGroup