openhwfoundation/core-v-verif #1031: debug_test hangs under dsim

MikeOpenHWGroup, 2021-11-22:
Running the `debug_test` on the `cv32e40x/dev` branch hangs when simulating with Metrics `DSIM`.  As far as I am aware, this behaviour does not occur on any other branch of any other core and does not occur with Cadence Xcelium.

At this time, I do not suspect an RTL issue.

### Steps to Reproduce
```
$ git clone -b cv32e40x/dev https://github.com/openhwgroup/core-v-verif.git
$ cd core-v-verif/cv32e40x/sim/uvmt
$ make test TEST=debug_test USER_RUN_FLAGS=+rand_stall_obi_disable SIMULATOR=dsim
```

MikeOpenHWGroup, 2021-11-23:
Note that I have merged in a hot patch to cv32e40x/dev (`c09848f`) to temporarily removed the debug_test from the Metric CI regression as the hanging test causes a false negative result and excessively long CI regressions.   The debug_test will be restored once this issue is resolved.

silabs-robin, 2022-02-04:
Hi. Are there any updates on this?
As I am un-commenting back in some 40x "rel_check" tests now, I noticed that "debug_test" is still commented out of 40x "ci_check".
Did someone from Metrics have a look at it?

MikeOpenHWGroup, 2022-02-04:
Sorry @silabs-robin, I have not had a look at this in a while.  I am working with Metrics on a simulation-runtime performance issue that I suspect may be related to this.  Once a new docker image for dsim is installed to resolve the performance issue I will look into this specific issue again.