openhwfoundation/core-v-verif #1045: install latest GCC compiler on OpenHW Metrics cloud

strichmo, 2021-12-03:
### Task Outcome
The Metrics cloud is updated with an installation of a top-of-tree gcc that provides full basic support for the 1.0.0 Zba, Zbb, Zbc, and Zbs extensions.  The minimal level of support required is that the assembler and the binutils support all opcode pnemonics for the above extensions supported by the cv32e40x and cv32e40s.

When the tool is installed, the cv_regress defaults for the cv32e40x and cv32e40s should point to the GNU toolchain as its default.  This should be introduced via a PR to the dev branches for both cores.

### Background information
The current _default_ toolchain for test development on CORE-V cores is the COREV toolchain provided by Embecosm.  This toolchain is a gcc-based toolchain that incorporates an up-to-date version of gcc with optimizations for COREV cores (e.g. PULP).  However maintaining the top-of-tree gcc to COREV is not a trivial process and thus the RISCV updates to gcc will lag in the COREV toolchain.  The Zb* extensions need to be supported now to test these instructions in the CV32E40X, as the instructions are already implemented.

The current top-of-tree gcc has the required level of support to facilitate random tests and some optimization in directed tests.
The earliest that the COREV toolchain may be updated with these changes is late Q1 of 2022.

### Location Information
The toolchain install directory of the Metrics cloud.

### Completion Criteria
Zb extension tests are added to ci_check/rel_check and these pass in the Metrics cloud.

MikeOpenHWGroup, 2021-12-03:
Hi @strichmo and @aimeepsutton, I have opened service ticket #1155 with Metrics for this task.

silabs-hfegran, 2022-01-04:
@MikeOpenHWGroup is this issue resolved?

MikeOpenHWGroup, 2022-01-04:
Yes, indeed.

closed 2022-01-04 by MikeOpenHWGroup