openhwfoundation/core-v-verif #1046: Potential bug in uvma_rvfi_instr_mon  (nmi/interrupt)

silabs-robin, 2021-12-06:
`uvma_rvfi_instr_mon` has [logic for detecting nmi and interrupts](https://github.com/openhwgroup/core-v-verif/pull/1044/files#diff-dc01f2b99b7d05d5d5f7840681f3c6e07704b2ce74c15f5eda2796551e155f6cL192), and am not sure it is right.

This concerns setting `mon_trn.insn_nmi = 1` and `mon_trn.insn_interrupt    = 1`.
Nmi is detected based on PC. but it should be more reliable to use rvfi and mcause? Some of the nmi detection should maybe be merged together with the interrupts detection too.
I think the code should rather be something like this:
```
         bit [XLEN-1:0] csr_mcause = mon_trn.csrs["mcause"].rdata;
         if (mon_trn.intr && csr_mcause[31]) begin
            mon_trn.insn_interrupt = 1;
            mon_trn.insn_interrupt_id = { 1'b0, csr_mcause[XLEN-2:0] };

            if (mcause[30:0] inside {128, 129}) begin
               mon_trn.insn_nmi = 1;
            end
         end
```

strichmo, 2021-12-06:
Yes, @silabs-robin this is planned to be updated based on the new multi-bit IRQ trap.  I will assign to myself and address today.

strichmo, 2021-12-07:
The issue should be cleaned up and addressed in this PR: https://github.com/openhwgroup/core-v-verif/pull/1051

strichmo, 2021-12-09:
Closing this issue as the PR was merged.

closed 2021-12-09 by strichmo