openhwfoundation/core-v-verif #1052: Need a simple hello world guide

silabs-robin, 2021-12-07:
There is a problem that setting up the environment is complicated and we don't have a minimal example.

Currently the main README's "Getting Started" has a link to what OpenHW is and to the verif strategy document which is too technical for someone looking for a simple hello world.
I see some /dev branches have added 40P and A6 links, and that might be sufficient (if 40x/s is similarly updated). But the readmes you find in `mk/` and `cv32e40x/sim/uvmt/` contain far too much information.

To make collaboration easier with either newcomers or 3rd parties, I think we should have a very minimal guide for running hello world.
It is also not clear which of the environment variables are essential for this, and which are extraneous.

**Example:**

To run hello world for e.g. cv32e40x, `cd` into `cv32e40x/sim/uvmt` and run `make test TEST=hello-world`.

The environment needs to be set up correctly. The following is a minimal example (read more in [todo link](...)).
```
setenv CV_CORE cv32e40x
setenv CV_SW_TOOLCHAIN <point to gcc>
setenv CV_SIMULATOR <set to "xrun" or some other simulator>
setenv USE_ISS NO
```

MikeOpenHWGroup, 2021-12-07:
Ha!   This is a case of either "great minds think alike" or "fools seldom differ".  The state of core-v-verif has been bothering me for a while and so I have been working on an update of the Verification Strategy and included a new [Quick Start](https://github.com/MikeOpenHWGroup/core-v-verif/blob/doc/docs/VerifStrat/source/quick_start.rst) chapter.   The idea is this would be pointed to directly from the top-level README.

This is still on a "doc" branch of my core-v-verif fork and its not yet ready to push up to the master, but your feedback would be welcome.

silabs-robin, 2021-12-17:
I have read through you "Quick Start" guide. It looks good.
The following are my thoughts about certain parts of the guide, but I have just dumped my thoughts straight out so I hope that they are still ok to read.

What is _"Lots of python3 plug-ins"_? This should either be described, reference some doc, or have an explanatory note by it?

This _"The above will clone a version of the CV32E40P"_ is a bit confusing start of that paragraph, since the commands above have just cloned _core-v-verif_.

Under _"Where is the RTL?"_, I think it should be emphasized that the RTL is not only cloned "as needed" but that it happens "automatically".

In _"$CORE_V_VERIF/cores/<core_name>"_ I get that "$CORE_V_VERIF" is supposed to mean wherever one cloned to, but does it cause confusion when one start to wonder if this is an official env variable one needs to set? And would it be better to just use "core-v-verif/cores/<core_name>" instead?

If I read _"The primary verification environments implemented in core-v-verif are all based on the UVM"_ I would wonder about whether the commands for "**core** testbench" shown above where part of this UVM bench or not, so maybe this should lead with some mention that the UVM benches and core benches are separate concepts?

I think maybe the _"SIMULATOR"_ variable should be _"CV_SIMULATOR"_? I don't have a complete overview, but it was my impression that the "CV_" prefixed variables were intended to be "public" (as in private/public in oop)?

There should be a link to [core-v-verif/mk/uvmt/README.md](https://github.com/openhwgroup/core-v-verif/blob/master/mk/uvmt/README.md#corev-environment-variables)'s "COREV Environment Variables" with an explanation that required environment variables are described there?
Actually, I am not even sure if any of the commands shown for running tests would even run if these env variables aren't set up first? I think maybe this should be made clear, and preferably early in the text.

The diff of the "doc" branch itself looks good too, based on skimming through it.

MikeOpenHWGroup, 2021-12-17:
Thanks for the feedback @silabs-robin, it is much appreciated.   I've pushed in some updates that should address your comments.

silabs-robin, 2021-12-21:
@MikeOpenHWGroup the getting started doc looks mostly good to me now.
I can't see that required environment variables are mentioned(?), but there is a link to it under the "Supported Simulators" section.
Other than that I think it is good.

MikeOpenHWGroup, 2021-12-21:
> I can't see that required environment variables are mentioned(?), but there is a link to it under the "Supported Simulators" section.

Good observation.  If the reader follows all the links to all the READMEs in the order they are provided in the Quick Start chapter, they will see the proper instructions for setting up the ENV vars.  I'll add a few words to encourage the reader to look at those READMEs (without violating the "DRY" principle of documentation: "don't repeat yourself").

MikeOpenHWGroup, 2021-12-21:
> I can't see that required environment variables are mentioned(?), but there is a link to it under the "Supported Simulators" section.

Good observation.  If the reader follows all the links to all the READMEs in the order they are provided in the Quick Start chapter, they will see the proper instructions for setting up the ENV vars.  I've added a few words to encourage the reader to look at those READMEs (without violating the "DRY" principle of documentation: "don't repeat yourself").

silabs-robin, 2021-12-27:
It looks good to me 👍

MikeOpenHWGroup, 2021-12-27:
Thanks for the reviews and feedback Robin.  I'll keep this issue open until the uodates to the Quick Start chapter are merged onto core-v-verif's main branch.

silabs-hfegran, 2022-03-03:
@MikeOpenHWGroup I presume this can be closed now?

MikeOpenHWGroup, 2022-03-04:
Ah, yes it can be.

closed 2022-03-04 by MikeOpenHWGroup